Atmosphere-NX/Atmosphere #1983: Switch Lite black screen at lock screen

dezem, 2022-11-06:
### What's the issue you encountered?

Use newest ams release 1.4.0 on my Switch Lite. Then updated from firmware 14.1.2 to 15.0.1. 
Boot with fusee.bin chainloaded with hekate. 
I often get blackscreen after the Nintendo Logo. At the moment where the lock screen should pop up. There is only blackscreen with turned on backlight. I musst hold power off for 10 seconds to hard reset the Switch and try it again. 

Sometimes it works and boots normaly.

If i downgrade back to 14.1.2 the Switch boots normaly each time. No more blackscreens. 

### Crash Report

N/A

### System Firmware Version

15.0.1

### Environment?

- Hekate as bootloader to chainload fusee.bin
- Latest official release
- No Homebrew
- emuMMC as partiton
- used clean SD with fat32

SciresM, 2022-11-06:
Screenshot of /atmosphere/contents?

dezem, 2022-11-06:
no contents used

[image]
[image]

SciresM, 2022-11-06:
Hm. If this is happening with emummc, could I ask what happens if you make a new emummc on 15.0.0 by cloning a 15.0.0 sysmmc, rather than upgrading an older emummc?

I know that's a hassle, but I'm not sure what else to try.

dezem, 2022-11-06:
My sysmmc is already on 15.0.1, so the fresh created emummc is the clone of the 15.0.1.

I looked at the erpt reports and there are created SystemPhysicalMemoryPeak files.
I attatched one of them, maybe you have an idea. My serial number is censored in that file.

[0c31a5d0-dc71-8349-7f76-d7cb678e6c05.txt](https://github.com/Atmosphere-NX/Atmosphere/files/9945820/0c31a5d0-dc71-8349-7f76-d7cb678e6c05.txt)

favoulos, 2022-11-10:
I have the same problem too, i'm on firmware 15.0.0, erista unit, i chainload fusee.bin with hekate. Sometimes the switch boot normally, sometimes i have black screen after the atmosphere and switch logo. I also deleted the contents folder for compatibility issues but nothing change

melquimachado, 2022-11-11:
Same issue here, OLED switch (Samsung NAND).

Latest Hekate 5.9.0, boot from fusee or Hekate fss0.
Clean Atmosphere 1.4.0 folder, without plugins, etc.

Cards tested:

SanDisk Ultra 512GB
Samsung Evo Plus 128GB
Kingston Canvas Select Plus 64GB

If EMUNAND as partition (resized or not, with all cards), black screen after Nintendo logo several times.

If EMUNAND as files (resized or not, with all cards) = all OK, the issue not occur. 

15.0.0, 15.0.1 = with EMUNAND as partition, resized or not, built from zero or updating with Daybreak = black screen issue

14.1.2 = all OK (files or partition)

Conclusion, problem with emunand as PARTITION with atmosphere 1.4.0 and 15.0.0 firmware.

melquimachado, 2022-11-11:
> Hm. If this is happening with emummc, could I ask what happens if you make a new emummc on 15.0.0 by cloning a 15.0.0 sysmmc, rather than upgrading an older emummc?
> 
> I know that's a hassle, but I'm not sure what else to try.

Same issue here, OLED switch (Samsung NAND).

Latest Hekate 5.9.0, boot from fusee or Hekate fss0.
Clean Atmosphere 1.4.0 folder, without plugins, etc.

Cards tested:

SanDisk Ultra 512GB
Samsung Evo Plus 128GB
Kingston Canvas Select Plus 64GB

15.0.0, 15.0.1 (built from zero or updating with Daybreak):

If EMUNAND as partition (resized or not, with all cards)
= black screen after Nintendo logo several times.

If EMUNAND as files (resized or not, with all cards)
= all OK, the issue not occur.

14.1.2:

all OK (files or partition)

Conclusion, problem with emunand as PARTITION with atmosphere 1.4.0 and 15.0.0 or 15.0.1 firmware.

TTghostsemir, 2022-11-15:
I have the Same Problem after the Switch Logo is Blackscreen I Must hold Power for 10-15 sec and than Works (Aula (Mariko) unit) I Hope the Issue is soon fixed! Firmware is 15.0.1

sepehrsb, 2022-11-17:
I've been having this issue with multiple modded OLEDs since the beginning when I started on ~14.0.0. Bug has probably been there since before that but I wouldn't know. Seems to happen with 512gb SD cards and emummc partition. Haven't seen the bug using a 256gb SD card. Sometimes it happens from time to time and other times it happens every other reboot. Force shutdown and rebooting always brings it back to working condition, the bug never happens twice in a row.

DarrenWarrenV, 2022-11-19:
Same problem here with OLED + hwfly. Literally every second boot I have to power it down, otherwise I'm getting a black screen. 15.0.0 + 15.0.1, on 14.1.2 everything's fine.

favoulos, 2022-11-23:
It seems the problem occurs when i restart the console from the menu. If I reboot via reboot to payload (to Hekate), I've had 100% success so far. Can you try it too?

SciresM, 2022-11-23:
@favoulos what you're saying doesn't make sense. Rebooting from menu takes me to hekate, on my Mariko devices.

I have also never had this problem.

sepehrsb, 2022-11-23:
@SciresM I'm now having this problem on my 256gb sd card modded switch OLED as well as my 512gb sd card modded switch OLED. SD card was formatted to fat32 with fat32gui and emunand partition. I've been getting this problem since 14.0 FW.

SciresM, 2022-11-23:
Yeah, there's kind of nothing I can do about it. I have literally never reproduced this even once, on 5 different units.

bas1977gld, 2022-11-24:
I don´t know if this is related, but since i updated atmosphere to 14.0 and switch update to 15.0.0 on both emunand and sysnand, my switch wont let me use sleep mode.
If i play with it on both sides it works normal, but when it turns in sleep mode, i have to hold the power button for 15 seconds to get it  to restart.
When i go to lakka and use the sleep mode, it does work.
So the problem is also black screen, just not with lakka.

favoulos, 2022-11-24:
> @favoulos what you're saying doesn't make sense. Rebooting from menu takes me to hekate, on my Mariko devices.
> 

I agree with you, hekate chainload fusee.bin but after the switch logo then i have a black screen 50% of the time.
If I reboot via reboot to payload (and then hekate starts and completes the chainload), I've had 100% success so far, instead

it's like _restarting from the HOS menu_ results in a black screen most of the time

if you want, I can attach the videos of the problem

favoulos, 2022-11-24:
my device is erista, atmo 1.4.0 fw 15.0.1

i cannot upload the videos, they are too big for github, can we continue on discord?

favoulos, 2022-11-24:
my device is erista, atmo 1.4.0 fw 15.0.1

first system reboot OK
https://user-images.githubusercontent.com/53868900/203819060-2e809637-3773-4c50-9ccc-10e04fcb4de9.mp4

second system reboot black screen
https://user-images.githubusercontent.com/53868900/203819103-4012cf58-2a01-43aa-ac00-76a6160d0bd2.mp4

first reboot with reboot_to_payload ok
https://user-images.githubusercontent.com/53868900/203819139-fbdfda80-11f6-4b50-82d6-23baa6a13e29.mp4

second reboot with reboot_to_payload ok
https://user-images.githubusercontent.com/53868900/203819174-22f924ba-7cf7-4ef5-8fa7-64f82dd28d61.mp4

DarrenWarrenV, 2022-11-25:
FOR EVERYONE HAVING THIS PROBLEM: go with a file based emuMMC, it solved ALL my problems. Even problem with black screen when charging in a powered off state! I know that it will be a headache, but it works.

sepehrsb, 2022-11-25:
Not an option for most of us due to limitations of file based emunand.

SciresM, 2022-11-25:
I don't actually think file based would solve the problem, because this seems like sd card hardware issues, but,

"limitations" what? I am not aware of any circumstance under which one can use a partition but not a file.

sepehrsb, 2022-11-25:
I was under the impression that you cannot move games between SD card and
file based emunand using the consoles system settings to move games between
the two. Also I thought file based was less stable and slightly slower in
performance.

SciresM, 2022-11-25:
The transfer issue may yet be legitimate -- I half remember something along those lines, though that would be a question for m4xw and not for me.

The performance thing has not been true in a long while -- file based will be as performant as partition based, and may be faster or slower depending on the specific card.

SciresM, 2022-11-25:
(It is definitely not less stable)

DarrenWarrenV, 2022-11-25:
I have a brand new 1TB sdcard, it's completely legit so that's not a problem with the card. But as I'm saying, everything is fixed now! I'll try to setup a new emummc on a 256GB card that I got with my Switch later on and upgrade it from 14.1.2 to 15.0.1 just to check if the problem exists. I checked my 1TB card for corruption, errors etc

DarrenWarrenV, 2022-11-25:
> I was under the impression that you cannot move games between SD card and file based emunand using the consoles system settings to move games between the two. Also I thought file based was less stable and slightly slower in performance.
> […](#)

I'll never use this option and if this is the only problem then I don't know. Maybe just try file based emummc if it helps?

DarrenWarrenV, 2022-11-25:
So, I made a brand new emuMMC, updated to 15.0.1 and the same thing 50% boot ratio. Well, I ran out of ideas. File based is the way to go for people having problems it seems. We're the chosen ones

sepehrsb, 2022-12-05:
Could my hekate.ini have anything to do with it?

[config]
autoboot=0
autoboot_list=0
bootwait=0
backlight=89
autohosoff=2
autonogc=0
updater2p=1
bootprotect=0

{--- EMUNAND Custom Firmware ---}
[CFW (EMUMMC)]
fss0=atmosphere/package3
kip1patch=nosigchk
emummcforce=1
icon=bootloader/res/emummc.bmp
{}

{--- Official Firmware ---}
[OFW (SYSNAND)]
fss0=atmosphere/package3
stock=1
emummc_force_disable=1
icon=bootloader/res/stock.bmp
{}

sepehrsb, 2022-12-08:
@SciresM could you confirm the hekate ini above? Thanks.

Also I've noticed with the atmosphere 1.4 build I've been using with 15.0.1 and creating an emunand with SD Partition, when I launch emunand it does the normal boot logos and then boots into a black screen everytime, it's not even 50/50 for me its everytime boot into black screen. Been forced to use SD File instead until this gets fixed as it has no problems other than slower boot times.

pingouin84, 2022-12-12:
same, 0/100 with emmc and sd file

SciresM, 2022-12-12:
I have still never reproduced this even once.

dezem, 2022-12-15:
Im staying on FW 14.1.2 with FW 15.X i never got it working.

SciresM, 2022-12-15:
Since this issue was created, I have formatted and created partition emummcs on three SD cards, and on both erista/mariko, never reproduced.

I do think this seems more like an emummc issue than an atmosphere issue, though, if it's only happening on SD-partition emummc for people reporting this.

avantearpeggio, 2022-12-22:
Just adding a data point:

Erista launch switch on - FW 15.0.1 using w/ Hekate 6.0.0 and Atmosphere 1.4.0

Can confirm that using the sd-partition method consistently hung after the Atmosphere -> Nintendo Switch logo screens. Switching to the SD File boot and it worked fine.

Hassas99, 2022-12-28:
I had more than a dozen Switch to mess around with, and I only witnessed this happens on one Switch in the past, so it's not something new.

It also started happening on my brand new OLED with the latest Atmosphere, though this time around is more persistent. Never used eMMC SD File in any of the devices.

favoulos, 2023-01-07:
Maybe i found a solution!
I noticed that I could no longer boot my original sysnand (3.0.2) through hekate's "stock" option, it stuck at the Nintendo logo. Talking to CTCaer we found a solution, but the strange thing is that now my emunand has a 100% bootrate also! So maybe it was a hekate problem? Tell me if I can post the fixed version of hekate here, to let other users with the same problem try it

bas1977gld, 2023-01-07:
I also am thinking it is an hekate problem. I noticed that my problems react different when directly boosting though hekate then if i go thought change payload, direct i cant sleep mode in lakka through change payload i can use sleep mode in lakka

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

Hassas99, 2023-01-10:
> Maybe i found a solution! I noticed that I could no longer boot my original sysnand (3.0.2) through hekate's "stock" option, it stuck at the Nintendo logo. Talking to CTCaer we found a solution, but [...]

This issue isn't new. I think the problem is more to do with the way SD is partitioned than hekate.
https://github.com/CTCaer/hekate/issues/172

mdx0504, 2023-01-12:
I think i fix it ! Boot Hekate -> go to Options -> Turn off (Auto HOS Power Off) and Turn off (Update Reboot 2 Payload).
I do this and Boot Atmosphere now it works fine :)

Lumince, 2023-01-29:
I can confirm that this is happening on the latest release with my chipped OLED with a file based emummc. This was even happening on my v1 while booting sysmmc. Still cant get them to boot passed the nintendo logo.

Lumince, 2023-01-30:
> You can try this version of Hekate, made by CTCaer. Rename it in update.bin and paste it in the bootloader folder. Let me know if it solves your problem
> 
> [CTCaer/hekate#815 (comment)](https://github.com/CTCaer/hekate/issues/815#issuecomment-1374523108)

Thank you I will test this!

Lamcza, 2023-01-31:
> > You can try this version of Hekate, made by CTCaer. Rename it in update.bin and paste it in the bootloader folder. Let me know if it solves your problem
> > [CTCaer/hekate#815 (comment)](https://github.com/CTCaer/hekate/issues/815#issuecomment-1374523108)
> 
> Thank you I will test this!

Do you use this sys module https://github.com/cathery/sys-ftpd ?
https://github.com/cathery/sys-ftpd/issues/22 some guys have similar issues. 
I have the same problem on my v1 when i set it to start on boot and only on 512sd and above sd cards. 

Ps. That fix does nothing for me.

impeeza, 2023-02-01:
Any to do with size - model of sd card?

Which one are falling?

dezem, 2023-02-05:
The fixed hekate build nothing change for me
I already tried it, for some time ago. 
https://github.com/CTCaer/hekate/issues/815#issuecomment-1374523108

My used SD cards for tests
Samsung Evo Plus 256GB and 512GB
Sandisk Ultra 512GB
Sandisk Extrem 1TB

With Firmware 15.1.0, first startup blackscreen. 
Mostly it works on the second boot. After hard powered off. Some times a third try needed. 

Im staying on firmware 14.1.2, because on this firmware i never got blackscreen isues.
And i dont use any custom content, like sysftp or any other!

Dav4555, 2023-02-07:
I have the same Problem. Have OFW 15.0.1 and if I create a partition based emummc then I stuck in black screen after the switch logo. Created a sdfile based emummc and I could boot into HOS. That's very strange.

masagrator, 2023-02-24:
> Is there a way to migrate to file-based from partition-based EmuMMC without losing data? I've tried looking around but only found guide for migrating to different SD card. Thank you in advance!

Issues is not an user support channel and this question relates to Hekate, not to Atmosphere.

mcwoot1337, 2023-03-10:
I've also been experiencing this issue. The black screen is consistently every other boot.

I've tested with two different SD cards (256GB and 1TB, both Sandisk) neither of which have anything in the contents folder. On my system, I've had it happen in both emuMMC and sysMMC. I'm currently in the process of setting up a file based emuMMC to determine if the partitioning is causing an issue.

Aula with 15.0.1 on sys and emu
Same results on both Atmosphere 1.4.1 and 1.50pre.
Hekate 6.0.2 with the below config

[Atmosphere emuMMC]
payload=bootloader/payloads/fusee.bin
icon=bootloader/res/emummc.bmp

[Atmosphere sysMMC]
fss0=atmosphere/package3
emummc_force_disable=1
icon=bootloader/res/sysnand.bmp

Please let me know if there is anything else that I can share.

SciresM, 2023-03-10:
@mcwoot1337 sanity check: you say it happens exactly every other boot for you?

can you confirm that result occurs booting using fusee instead of hekate, and exclusively into sysmmc with an empty atmosphere/contents directory?

mcwoot1337, 2023-03-10:
@SciresM Yes, from a cold boot with either emu or sys (splash, logo and then black).

I attempted to boot fusee as my payload with both sysmmc and emummc and it goes straight to a black screen on every boot (no splash or logo). I'm short on time today, but I could give this another try tomorrow.

SciresM, 2023-03-10:
To be sure: have you disabled emummc in hekate's menu before booting fusee? You need to use the change emummc menu.

mcwoot1337, 2023-03-10:
I manually changed enabled from 1 to 0 in emummc.ini. I will attempt disabling from hekate.

SciresM, 2023-03-10:
These problems are also often resolved by making a fresh mmc image, especially if sysmmc fail (which should never occur and isn't what anyone else in this issue reports). I would try taking a backup and then following an unbrick guide to flash a factory image.

mcwoot1337, 2023-03-10:
Will do, thanks @SciresM!

ch4d1, 2023-03-12:
I have also tried to recreate the error on my Aula Switch. Without success. However, it would be interesting to know if it is perhaps a faulty SD card port. This could be determined in Hekate via Console Info -> microSD. Look out for errors or poor read rates.

mcwoot1337, 2023-03-12:
I wanted to share an update from my further troubleshooting in hopes that this may help someone else. As SciresM had mentioned, my problem is very likely unrelated to this issue. If there is a need, I'd be happy to open a new issue.

I started noticing that about 20 seconds after initiating a power off from HOS, that the chip was glitching and launching hekate. When attempting to launch atmosphere from hekate in this state, it always leads to a failed boot (splash, logo, black). In my case the black screen is an actual power off, which would explain my 50% success rate. Any boot after a true power off is 100% successful.

The band-aid solution was to set autohosoff=2 in hekate_ipl.

favoulos, 2023-03-12:
Have you tried setting autohosoff=0 instead? This way when you shut down from HOS, the system shuts down completely right away.

Zac5teeve, 2023-03-12:
Original version of the switch not waking up/black screen "when put to sleep in dock" 100% of times

Only way to get out of black screen is by hard rebooting holding volume+power until wake and boots in system "not atmosphere"

Hassas99, 2023-03-13:
> I started noticing that about 20 seconds after initiating a power off from HOS, that the chip was glitching and launching hekate. When attempting to launch atmosphere from hekate in this state, it [...]
This is normal behavior of the chip and fixes nothing.

1khanal1, 2023-03-26:
Any progress on this issue? Having the same issues with the same build using Hekate as a bootloader and then using fusee. 

The issue could be related to Hekate 6.0 onwards but I'm parroting terms used by others here I am super illiterate relating to homebrew functionality.

brettsmods, 2023-03-28:
> Any progress on this issue of just getting stuck on a black screen after updating to 1.5.0/1.5.1? Having the same issues updating AMS to 1.5.0 and 1.5.1 with the same build using Hekate as a bootloader and then using fusee.
> 
> Had to downgrade to AMS 1.4.0. Currently running Hekate 6.02 with AMS 1.4.0 on system firmware 15.0.1
> 
> Would AMS 1.5.1 work if I update the system software to 16.0.1?

Exact same issue here. Was going to try updating to 16.0.1 next after a backup, so I'll let you know.

closed 2025-04-03 by SciresM