Atmosphere-NX/Atmosphere #1984: Crash While Running Error Code 2110 0350

AnthonieZ87, 2022-11-06:
## Bug Report
[image]
I myself and 2 others I know of have been experiencing a crash with the same error.
Nintendo switch error codes for 2110 start at 2110-1000 so I can't find any info related to what is causing it.

Seems to occur while running pokemon sword /shield perhaps while integrated with pkhex using the live hex option
but i can't say for certain.  It's happened to me once and honestly, I don't think I was using pkhex at the time but is possible. another person I know who received this error and has gotten it a bunch of times now but hasn't mentioned what they were doing exactly that led to it. I just can't seem to find anything related to that error code and we have been looking everywhere

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
<script src="https://gist.github.com/AnthonieZ87/fc35b1b6d6ff199a4e9a9db5a44e6d4a.js"></script>
<script src="https://gist.github.com/AnthonieZ87/f9db28abfb2d2ddb5fedb28f9616c35f.js"></script>

### System Firmware Version

15.0.1</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate latest build
- Official release or unofficial build: official [hekate v5.9.0](https://github.com/CTCaer/hekate/releases/tag/v5.9.0)
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND: SysNAND
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

Masamune3210, 2022-11-06:
Please go back and fill out the rest of the issue template. The information is asked for a reason

Masamune3210, 2022-11-06:
Also, you didnt wait long enough for the crash report to upload, please reupload it

AnthonieZ87, 2022-11-09:
OK I still think I messed up but hoped that helps

Masamune3210, 2022-11-09:
Just upload the error report as a file, don't stick it in a gist

Masamune3210, 2022-11-09:
Hmm, 010000000000002F is npns, the service that handles Notifications

AnthonieZ87, 2022-11-10:
> Just upload the error report as a file, don't stick it in a gist

[image]

I did, it’s literally the last line lol

Masamune3210, 2022-11-10:
You can zip the file and upload it that way

SciresM, 2022-12-04:
`nifm::ResultResourceNotFound()` -- this isn't an atmosphere issue, it will be arising from whatever custom network stuff you're using.

closed 2022-12-04 by SciresM