Atmosphere-NX/Atmosphere #1985: Panic occurred while running Atmosphere: Title ID 010000000000002B

chrisdaun, 2022-11-07:
## Bug Report

### What's the issue you encountered?
emuMMC will no loner load into the Switch OS
Atmosphere crashes with a panic error message shortly after the Nintendo Switch splash screen.

Title ID: 010000000000002B
Error: std::abort (0xFFE)

No changes to Atmosphere were made

### How can the issue be reproduced?
The issue can be reproduced by booting into hekate and then launching emuMMC with fusee.bin or package3.bin
As soon as the Nintendo Switch logo fades out, the error message is displayed

### Crash Report
https://gist.github.com/chrisdaun/9d9e70cabf940304f6dca7bbc69d959b

### System Firmware Version
14.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate 5.7.2
- Official release or unofficial build: Official
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Daybreak, reboot to payload
- EmuMMC or SysNAND: partition-based EmuMMC

### Additional context?
I did try the suggestion in this bug by @Adubbz  https://github.com/Atmosphere-NX/Atmosphere/issues/1256
Deleting 8000000000000050 and 8000000000000052 from the save folder in the SYSTEM partition but this did not resolve my issue.

BeckysFootSlave, 2022-11-07:
The message comes from "erpt"!

Does it work if you delete the "erpt_reports" folder inside atmosphere folder on SD?

chrisdaun, 2022-11-07:
Thanks for the idea @BeckysFootSlave but removing the erpt_reports folder didn't fix the issue.
Upon first boot after removing the folder, the system gave the same panic message and a new erpt_reports folder was created with a new report in it.

BeckysFootSlave, 2022-11-07:
@chrisdaun sad to hear that!
Some crashes happened because that folder contained to many files!
Really thought it would help to delete it to get under 1000 files (most of those crashes started at 1000 and more)

Search for "fatal error 010000000000002B" maybe you find the right solution!

chrisdaun, 2022-11-07:
Thanks @BeckysFootSlave most of the info I found around the internet recommend removing the 80000000000000d1 save file from the SYSTEM partition but that didn't work either. Hoping the attached atmosphere crash log will help the devs help me :)

BeckysFootSlave, 2022-11-07:
@chrisdaun I really hope you can solve this!
Will take a eye on this!

Maglata, 2022-11-20:
Hey did you manage to find a solution, since I have the same problem

chrisdaun, 2022-11-20:
@Maglata no solution yet unfortunately

SciresM, 2022-12-04:
@chrisdaun Please upload the report somewhere else. It is corrupted, because it is a binary file that you posted as a text file.

chrisdaun, 2022-12-05:
@SciresM thanks for the reply, I ended up blowing away the SD card and setting up a new emuMMC. If I hit the issue again, I'll open another issue. Many thanks for the awesome work.

closed 2022-12-05 by chrisdaun

pintcat, 2022-12-24:
I'm having the same issue right now (the exact same error message when booting Atmosphere). Tried all the suggestions, but nothing worked. I thought I might find some help here before I set up everything from scratch. Attached my to this message you can find my error report. Any help would be appreciated.
[report_0865e61e.zip](https://github.com/Atmosphere-NX/Atmosphere/files/10299079/report_0865e61e.zip)

ajchen01, 2024-01-08:
I solved the problem using the method in this video. Just follow its steps and you'll be okay, except for one thing: you need to delete the file named '80000000000000d1'. 
https://www.youtube.com/watch?v=I0cFkxxOss4

eodrambo, 2024-12-03:
I just had a similar issue, I tied to follow the video @ajchen01 posted, however it was in another language, thankfully the video is a rough interpolation of the article below. I used a backup I had, but it seems it's possible to grab the relevant files from the system nand (this is what the video does based on my rough understanding of the auto translated subtitles) Hopefully this helps if someone else runs into this issue in the future.

https://alvinr.ca/2023/05/21/fixing-a-fatal-error-occured-when-running-atmosphere-title-id-010000000000002b/