Atmosphere-NX/Atmosphere #1987: atmosphere creating files in erpt_reports even when no crashes are happening

NekoNoor, 2022-11-09:
### What's the issue you encountered?

when booted into atmosphere a lot of files are created in the erpt_reports folder even when i am not doing anything

### How can the issue be reproduced?

launch atmosphere and wait for a couple minutes and check the sd card erpt_reports folder, there will be a lot of files causing it to hit the FAT limit of 16383 files in that folder and making the switch completely unusable (games load noticably slow and homebrew works extremely badly)

### Crash Report

N/A

### System Firmware Version

15.0.1

### Environment?

- hekate (launching atmosphere using fusee.bin)
- Official release version 1.4.0
- Do you have additional kips or sysmodules you're loading: N/A
- Homebrew software installed: N/A
- SysNAND

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

Masamune3210, 2022-11-09:
Post a couple of the reports, we can't know what is going on if we don't have examples

NekoNoor, 2022-11-09:
[erpt_reports.zip](https://github.com/Atmosphere-NX/Atmosphere/files/9973222/erpt_reports.zip)

Masamune3210, 2022-11-09:
Do you have a account linked to a Ninty ID by chance?

NekoNoor, 2022-11-09:
yes my main account is linked to a nintendo ID

Masamune3210, 2022-11-09:
The error code that appears in the logs, 2234-4508, is account has been banned from using online services

NekoNoor, 2022-11-09:
i unlinked the account from nintendo and it seems to have fixed it, not sure why it would generate an error every few seconds if it's banned though

Masamune3210, 2022-11-09:
No clue, never had to deal with it myself so I'm not entirely sure of how a banned switch normally behaves

Masamune3210, 2022-11-09:
Might leave this open so someone can look more into it just in case atmosphere is behaving incorrectly

masagrator, 2022-11-09:
First of all erpt is not a sign of crash. Crash generates crash report, not error report.

Masamune3210, 2022-11-09:
Yeah it's not crashing, but it's definitely not happy either

NekoNoor, 2022-11-11:
After I unlinked my account it's no longer making all those error files, but now I can't play certain games that require a linked account.

My switch has been banned since 2018 so I'm not sure what changed that it started making all those errors. If I link my account using Linkalho the same thing happens as when it was linked to a real account like it has been since I got my switch.

loganm911, 2022-11-12:
ive been having this happen as well, also linked with linkalho. although for me it started right as the system started to nag about an available update. so it also happened on 15.0.0.

NekoNoor, 2022-11-12:
It doesn't seem like it, I left it for a while and hasn't made any new files.

NekoNoor, 2022-11-12:
I enabled prodinfo blanking (again) and it still makes the files, the only way to stop it is to unlink the accounts. But I had it linked for several years while being banned and never encountered this

Masamune3210, 2022-11-12:
are you sure its enabled? exosphere's config is the weird one, it goes on the root of the drive instead of the config folder

Masamune3210, 2022-11-12:
Check in Settings if your Serial Number for the console is different, that's the easiest way to see if its working or not

NekoNoor, 2022-11-12:
> are you sure its enabled? exosphere's config is the weird one, it goes on the root of the drive instead of the config folder

I checked and it shows the serial number as all zeroes

Masamune3210, 2022-11-12:
Can you upload a log from after you enabled blanking? Make sure its one of the new ones, easiest way would be to clear out the folder and then grab one

NekoNoor, 2022-11-12:
[erpt_reports.zip](https://github.com/Atmosphere-NX/Atmosphere/files/9996029/erpt_reports.zip)

Masamune3210, 2022-11-12:
2123-0011 is  Ninty servers refusing the connection because your switch cant prove its a legitimate console due to the blocked PRODINFO. Something is causing the system to repeatedly try to connect to Nintendo's servers. Maybe latest update changed the behavior of something like cloud saves?

@NekoNoor Error code is now 2155-8007 which is interesting, most of what I can find is saying that its relating to a secondary console trying to contact Nintendo to verify account details and licenses

Masamune3210, 2022-11-12:
Does putting the console into Airplane Mode stop the error spam?

NekoNoor, 2022-11-12:
> Does putting the console into Airplane Mode stop the error spam?

Yes

Masamune3210, 2022-11-12:
Did you happen to have another console set as primary?

NekoNoor, 2022-11-12:
> Did you happen to have another console set as primary?

Nope

Masamune3210, 2022-11-12:
Well, unless one of the actual devs wants to jump in, I am out of ideas other than taking a nand backup and then resetting your switch so it no longer has the legit account connected. Since the account is banned, you wouldnt be losing much from not having it connected anyway

hexkyz, 2022-11-15:
This issue is not related to Atmosphère, but rather to how Nintendo's servers work. A large number of error reports being generated is expected behavior since the console is indeed spamming a ton of errors when connecting.

closed 2022-11-15 by hexkyz

NekoNoor, 2022-11-15:
The problem isn't the errors itself but rather the files the errors create, which is a feature of atmosphere which you can't turn off, something like a toggle for saving erpt errors to sd would make it at least usable again like how it used to be.

Did Nintendo change this on their end? Because I didn't used to get this

hexkyz, 2022-11-15:
There was a recent(-ish) change in server checks yes. Some thought will be needed to decide on adding a setting for this. In the meantime, it should be possible to set the "erpt_reports" folder to read-only to prevent a crash from happening.

DraconicNEO, 2023-01-12:
> There was a recent(-ish) change in server checks yes. Some thought will be needed to decide on adding a setting for this. In the meantime, it should be possible to set the "erpt_reports" folder to read-only to prevent a crash from happening.

Sorry about the late Reply but I'm having the same issue and I want to ask How do I write protect the erpt folder, I tried in windows and linux and neither did anything (still get tons of reports in the folder) SD card is formatted as FAT32 if that helps.

zexu-xu, 2023-03-13:
Delete erpt_reports folder and create an empty file named as erpt_reports should do the trick.

rgardner168, 2023-06-21:
> Delete erpt_reports folder and create an empty file named as erpt_reports should do the trick.

doing this does fix the slow issues. however, it breaks other atmosphere features like DNS blocking on nintendo servers. if you do this and then try to connect to a nintendo server (e.g. try to 'update a game' or check for updates) atmosphere crashes.

SciresM, 2023-06-21:
deleting that folder will definitely not cause atmosphere crashes. it'll be something else in your setup.

rgardner168, 2023-06-21:
sorry should have been more specific on the error / crash. i think it is the creating of the 'empty' file named 'erpt_reports' that was suggested. simply deleting the folder but not creating a file named 'erpt_errors' seems to just recreate the folder once the switch is booted into atmosphere and the endless errors from the ban start back up and slow the system until it eventually crashes...

rursache, 2023-07-15:
this issue must be addressed. every 2 seconds or so another "log" file is written to the `erpt_reports` folder. not only its wasted performance but it also kills the sd card. also if i keep the switch on for a few hours, i'll be unable too boot (after a reboot or shutdown) back in atmosphere until i manually delete that cursed folder. how is this acceptable in any way or form?

Masamune3210, 2023-07-15:
Wow, its almost....like its not intended behavior or something...

rursache, 2023-07-15:
> Wow, its almost....like its not intended behavior or something...

why would preventing boot be a "intended behavior"? 'cos i sure can't boot once that folder gets filled constantly within the hours... i found kefir OS, a change to the default atmosphere with logs disabled. guess what. the folder now says empty and i can actually reboot my console without having to constantly get the sd card out to clean the folder.

liamwhite, 2023-07-21:
The fact that a continuous stream of error reports is generated means something is going wrong on your device. That could be due to many factors. Many are potentially caused by custom sysmodules, which you should remove and retest without (rename atmosphere/contents to atmosphere/contents.old) to ensure they are not the issue. Atmosphere is just the messenger.

DraconicNEO, 2023-07-22:
> The fact that a continuous stream of error reports is generated means something is going wrong on your device. That could be due to many factors. Many are potentially caused by custom sysmodules, [...]

2155-8007 means can't connect to Nintendo servers when there is internet access, it happens when you block Nintendo servers. So in essence we know the exact thing that is causing the error stream it's no mystery to us it's connection spam when the servers are blocked. The real problem here actually is the error stream because the logs are saved with no way to disable this functionality and they eventually hit the filesystem driver limit crashing the OS, again the problem isn't the errors, it's the logs that keep piling up. It's already been stated here in this discussion that it is caused by online server checks due to blocking Nintendo servers:

>There was a recent(-ish) change in server checks yes. Some thought will be needed to decide on adding a setting for this. In the meantime, it should be possible to set the "erpt_reports" folder to read-only to prevent a crash from happening.

DraconicNEO, 2023-07-22:
> The fact that a continuous stream of error reports is generated means something is going wrong on your device. That could be due to many factors. Many are potentially caused by custom sysmodules, [...]

@hexkyz Out of curiosity is a fix for this to disable erpt reports still being considered by the way?

ehx-v1, 2023-07-31:
Honestly, I don't think there's a way to fix it short of setting up a whole server infrastructure that replaces Nintendo's servers entirely (similar to Wiimmfi) and DNS rerouting all of the Switch's server requests to that. Which we'll probably all agree isn't something we can reasonably expect to be done within a few weeks. Even if the community felt comfortable setting up custom servers, it would take years of work.

DraconicNEO, 2023-07-31:
> Honestly, I don't think there's a way to fix it short of setting up a whole server infrastructure that replaces Nintendo's servers entirely (similar to Wiimmfi) and DNS rerouting all of the Switch's [...]

There's a simpler solution, allow people to disable erpt report logging entirely. See you're focusing too much on what the erpt reporting software is reporting but that's not actually the problem that's being caused here. The real problem is that Games and Software spam Nintendo servers when they can't connect and Atmosphere writes the error logs to the SD card eventually hitting the file limit and crashing Horizon, not to mention slowly killing the SD card due to write cycles.

I would love to have custom servers on the Switch as much as anyone but not be forced to use them as a band-aid fix for a problem which atmosphere created in the first place by forcing users to save logs even if they don't care about them. The solution to this problem is to allow users to opt out of erpt logging and just send them to /dev/null or the horizon OS equivalent.

Masamune3210, 2023-07-31:
They are focusing on the error because that is the correct thing to focus on lol. Erpts are written to the SD card for a reason, they are useful to fix errors. Disabling them entirely just because of one type of error is imo kind of overboard.

SciresM, 2023-07-31:
So, to be explicit: 

I do not want to add a mechanism that a user can toggle that allows disabling error reporting. I am flatly not willing to do this.

The inevitable result of doing this will be that people who make "packs"/distribute atmosphere alongside homebrew will disable error logging, and then people will reach out for support, and they won't have logs. That's clearly unacceptable to me.

Also, to be clear, consoles should not be logging thousands of errors, and I would also like to note that if atmosphere was not writing to them to the SD card, the official nintendo software would be writing them to the internal emmc, which seems even worse from a wear/tear perspective to me?

I would maybe be willing to consider not saving further reports if the user has a stupidly large amount present. Maybe...5000? I could see an argument for not writing reports to the SD card if there are a stupidly large number already present. This would, I guess, require the automatic cleanup stuff to be off to work.

DraconicNEO, 2023-08-01:
> I do not want to add a mechanism that a user can toggle that allows disabling error reporting. I am flatly not willing to do this.
>
> The inevitable result of doing this will be that people who make [...]

Not like they can't do it anyway, atmosphere is open source after all, I think there are a few that do it (no I won't link them here). 

> Also, to be clear, consoles should not be logging thousands of errors, and I would also like to note that if atmosphere was not writing to them to the SD card, the official nintendo software would be writing them to the internal emmc, which seems even worse from a wear/tear perspective to me?

I know it does, that's why I suggested something like destroying them entirely i.e. /dev/null (or a similar equivalent) approach where they aren't saved anywhere and are wiped from existence upon creation.

> I would maybe be willing to consider not saving further reports if the user has a stupidly large amount present. Maybe...5000? I could see an argument for not writing reports to the SD card if there [...]

That's a fair compromise if you don't want to disable it entirely. I mean the cleanup script doesn't really work if people don't reboot their consoles before hitting the driver limit most people keep their switches asleep and powered on because it's just more convenient with the tethered nature of the RCM exploit.

SciresM, 2023-08-01:
> Not like they can't do it anyway, atmosphere is open source after all, I think there are a few that do it (no I won't link them here).

Well, yes, part of being open source is that people can do whatever they want. That is their prerogative and I support people doing whatever they want. But if someone is using a pack, I can tell them "I don't provide support if you're using that", in a way that I can't if it's native to Atmosphere.

> I mean the cleanup script doesn't really work if people don't reboot their consoles before hitting the driver limit most people keep their switches asleep and powered on because it's just more convenient with the tethered nature of the RCM exploit.

I would like to reiterate one more time that I think this whole problem is complete nonsense. But, yes, not writing if there are already thousands of files, or if we've written 1000 files since the last reboot, or some other insanity metric seems not unreasonable to me.

SuperSpe, 2023-08-07:
> > Not like they can't do it anyway, atmosphere is open source after all, I think there are a few that do it (no I won't link them here).
> 
> Well, yes, part of being open source is that people can do whatever they want. That is their prerogative and I support people's right to do whatever they want. But if someone is using a pack, I can tell them "I don't provide support if you're using that", in a way that I can't if it's native to Atmosphere.
> 
> > I mean the cleanup script doesn't really work if people don't reboot their consoles before hitting the driver limit most people keep their switches asleep and powered on because it's just more convenient with the tethered nature of the RCM exploit.
> 
> I would like to reiterate one more time that I think this whole problem is complete nonsense. But, yes, not writing if there are already thousands of files, or if we've written 1000 files since the last reboot, or some other insanity metric seems not unreasonable to me.

I have at least 16k files inside that folder and on the boot the switch becomes very slow.
I want to know in my case why these reports are made and correct the errors.
How can I view them? They are not plain text...

DraconicNEO, 2023-08-07:
> I have at least 16k files inside that folder and on the boot the switch becomes very slow. I want to know in my case why these reports are made and correct the errors. How can I view them? They are [...]

Erpt reports are indeed plaintext and you can view them in a text viewer, that's how I figured out that they're all the same error related to Nintendo network.

DraconicNEO, 2023-08-08:
> > > I have at least 16k files inside that folder and on the boot the switch becomes very slow. I want to know in my case why these reports are made and correct the errors. How can I view them? They are not plain text...
> > 
> > 
> > Erpt reports are indeed plaintext and you can view them in a text viewer, that's how I figured out that they're all the same error related to Nintendo network.
> 
> strange. I search for the errore inside the zip and did not find any of this errors on single file. what I am missing?

If you open one of the report files it'll have the error code in it somewhere in the text as a four digit number with a dash followed by another four digit number (xxxx-xxxx) i.e. 2155-8007 which is when Nintendo servers cannot be contacted, (happens usually when people block the servers).

ehx-v1, 2023-08-08:
When I took a look at the errors in the ZIP above (since I deleted the reports that were on my own SD card), I saw a lot of binary symbols mixed into the file. Maybe the encoding is misrecognized?

SciresM, 2023-08-08:
Error report files are [messagepack](https://en.m.wikipedia.org/wiki/MessagePack) format.

anyway, can we please not spam this issue with comments? I will lock it if this continues. This isn't the right place for questions about how to decode report files.

jamjazz, 2023-08-26:
If I can make my modest contribution.
I have the same problem too.
I believe it started when I re-linked my user accounts with linkalho.
Maybe that's the problem...
If I unlink accounts, everything is back to normal.

SuperSpe, 2023-08-28:
> If I can give my modest experience to you.. I have the same problem too. I believe it started when I re-linked my user accounts with linkalho. Maybe that's the problem... If I unlink accounts, everything is back to normal.

I also had a suspicion it was really link halo, now I have the confirmation

ehx-v1, 2023-08-28:
So we're back to "Atmosphère is not responsible for this issue". I'd suggest reporting the problem to Linkhalo and then calling it a day.

NekoNoor, 2023-08-28:
it's also reproducible if you had an account linked without linkalho, so it isn't necessarily an issue with linkalho but an issue with having an account linked

jamjazz, 2023-08-28:
Yep.. same problem with Tinfoil fake-linked account pass.

NonaSuomy, 2023-09-10:
Had this issue as well when I linked the accounts with linkhalo It fiercely generated 10K erpt_reports in 1 night causing the system to slow to a crawl. Atmosphere should toss up a warning when this folder becomes polluted or fix why having a linked account causes this issue. Ideally, it doesn't seem like I needed a linked account for anything after unlinking them. It stopped generating the reports and everything worked fine again after purging the folder. The only reason I did it was I had a corrupted DLC that was tossing up Nintendo account-required message after reinstalling again without the DLC that message stopped as well.

ehx-v1, 2023-09-11:
> fix why having a linked account causes this issue

That's the whole Nintendo Network problem, seems like it tries to look up the linked account and fails.

> a corrupted DLC that was tossing up Nintendo account-required message

That kind of stuff can be forced to shut up with sys-tweak as described above, seems the much better option to me compared to setting up a linked account literally nothing for Singleplayer has any business needing.

springroll-xx, 2023-09-23:
> Had this issue as well when I linked the accounts with linkhalo It fiercely generated 10K erpt_reports in 1 night causing the system to slow to a crawl. Atmosphere should toss up a warning when this [...]

How does one know which DLC is corrupted that would toss up account-required message?

NonaSuomy, 2023-09-27:
> How does one know which DLC is corrupted that would toss up account-required message?

It only tosses up that error when you open that game so I would say delete each update/dlc with tinfoil update remove options until the error goes away then download fresh/working ones then add them back and see what happens.

lucasvm01, 2024-03-30:
Hello, I apologize if I should not ask about it here, and if not, please tell me where to.
I'm currently using FW 17.0.1 and AMS 1.6.2.

I was trying to play the Nintendo Switch Online games, which require a link to a Nintendo account, so I used Tinfoil to fake link an account and soon got this flood of errors on my atmosphere/content/erpt_reports folder.
I tried the sugestion from @Ammako:
>So I suggest: get sys-tweak, make a folder in atmosphere/contents for each of your games that require a linked account to run, and make a config.ini like this in it:
>```
>[override_nacp]
>startup_user_account=1
>```
>See if this works and if you're able to run all of your games with Nintendo account unlinked. This should hopefully avoid the issue of too many erpt reports being created, while still being able to play your games.

And it did not work on the Nintendo Switch Online games (all of them), so I tried it on the game they previously tried, Bloodstained: Ritual of the Night, and it did work.
So I'm guessing it doesn't work on the Nintendo Switch Online games specifically.

I would like to ask if anyone has found a solution for playing those games, without fake linking an account or downloading Retroarch or another third party emulator.