Atmosphere-NX/Atmosphere #1988: Error on Boot 2002-2634

Peacock1090, 2022-11-09:
## Bug Report

Updated Atmos to latest release while updating the Switch's SYS to latest firmware. For some reason (can't remember my reasoning) I then went to Daybreak and downgraded it to either 10.0.1 or 13.0.1. Tried to boot Atmosphere using the same files as the latest release and it keeps going on a loop of rebooting over and over until I hold down the power button. Tried to at least boot it without CFW to see if I can update it manually and get it working again but this even produces an error where I have to shut the system off completely.

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

1. Be a moron like me and update your Switch to the latest firmware, then downgrading it again even though it at least booted up with default Atmosphere settings.
2. ???
3. Profit
Haha remember that old meme gang? That's about all I remember.

### Crash Report

Atmosphère Crash Report (v1.7):
Result:                          0x149402 (2002-2634)

Process Info:
    Process Name:                ns

This is all it can produce before it reboots abruptly, corrupting the rest of the report. It goes by so fast I don't even have enough time to read it, but I remembered it did say something about title_id 010000000000001f, but I have nothing in the contents folder at the moment.

### System Firmware Version

Either 10.0.01 or 13.00.01. I have memory issues but it's probably the first one for some reason.

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  Official. Tried Hekate too to see if this would work but it didn't.
- Do you have additional kips or sysmodules you're loading:
Deleted them just to see if it can at least boot up and just add them again later. Nothing.
- Homebrew software installed: [ * ]
SysNAND. I don't have an Emulated one and didn't really make one since I don't have a use for it if that helps.
### Additional context?

Is there at least a way to boot it up via OFW and then updating it and seeing if it works again? Had to do this before when I messed up updating Atmos one time (yeah idk why this happens to me a lot) and that seemed to somehow fix the issue. I also don't have Mission Control installed and don't even know what that is really.

beyondmeat, 2022-11-22:
This is not a bug. You need support. Basically shutdown your switch, put your sdcard in the computer. Download the latest atmosphere and hetake. Extract the zips to the sdcard, overwrite files. While you do this, you can boot your switch w/o SD card and it will boot up stock. You can update stock FW through the sys menu.
Shutdown your switch, pop in the sdcard and boot with RCM. Now you can update the emunand firmware with aio-updater and daybreak. I recommend 15.0.0 instead of 15.0.1

Peacock1090, 2022-11-23:
I don't have access to my Switch and all atm but yeah something tells me this will fix the issue. I'll go ahead and close this and if I run into any issues I'll just follow your suggestions.

closed 2022-11-23 by Peacock1090