Atmosphere-NX/Atmosphere #1990: Crash 2002-0039 when checking a game blocked by exosphere

Reaster0, 2022-11-12:
## Bug Report

### What's the issue you encountered?

when i try to launch any game, and i'm connected to the internet, after the message saying "can't check the game" (obviously since exosphere stop the connection) the switch will crash,
[image]

[IMPORTANT] the issue will not reproduce if the switch is in airplane mode

### How can the issue be reproduced?

be connected to the internet, with a exosphere setup, try to load a game with sigpatches up to date, and the crash should appear just after the message saying "can't check the validity of the game" (or something)

### Crash Report

https://gist.github.com/Reaster0/b3042c38a13074f091aa4720b39a9e3b

### System Firmware Version

15.0.1|AMS 1.4.0|E

### Environment?

using fusee (latest) hekate (latest aka 5.9.0) ans latest sigpatches
running on EmuMMC (maybe the issue can be reproduce in SysNand, haven't check because my SysNand is clean so i haven't activate exosphere in it)
the EmuMMC is file based (i think?)

### Additional context?

here is my exosphere config if that's usefull

[exosphere]
debugmode=1
debugmode_user=0
disable_user_exception_handlers=0
enable_user_pmu_access=0
blank_prodinfo_sysmmc=0
blank_prodinfo_emummc=1
allow_writing_to_cal_sysmmc=0
log_port=0
log_baud_rate=115200
log_inverted=0

zemixe, 2022-11-12:
Piracy is not supported.

Reaster0, 2022-11-12:
the issue as nothing to do with piracy, the issue is caused in any program that you've installed as NSP (not only games)
and that exosphere is used to prevent your switch to be banned by that

masagrator, 2022-11-12:
installing anything as NSP is equal to committing into piracy

hexkyz, 2022-11-15:
Piracy is not supported.

closed 2022-11-15 by hexkyz