Atmosphere-NX/Atmosphere #1991: Error 524850534F53454D

genjiow2, 2022-11-14:
Trying to load Atmosphere I get an error 524850534F53454D
OFW 15.0.1
If you need report file,text me discord Lok#6218

masagrator, 2022-11-14:
Nobody will write to you on DM. 

Use issue template, fill it properly and upload reports here.

Your issue in current form is useless.

genjiow2, 2022-11-14:
> Nobody will write to you on DM.
> 
> Use issue template, fill it properly and upload reports here.
> 
> Your issue in current form is use
For some reason the file cannot be attached.

genjiow2, 2022-11-14:
> Nobody will write to you on DM.
> 
> Use issue template, fill it properly and upload reports here.
> 
> Your issue in current form is useless.

For some reason the file cannot be attached.

masagrator, 2022-11-14:
"For some reason" that is clearly explained?

[image]

Just zip it.

genjiow2, 2022-11-14:
> "Почему-то" это понятно объяснено?
> 
> [image]
> 
> Просто застегните его.

Thank you

hexkyz, 2022-11-15:
Please fill out the issue template.

closed 2022-11-15 by hexkyz