Atmosphere-NX/Atmosphere #1994: Error 524850534F53454D

genjiow2, 2022-11-15:
OFW 15.0.1
When I try to start Atmosphere, I get an error
Title ID:524850534F53454D
SD card is blank
Before that was SX OS
There are no mods to add memory or increase performance.
[report_000000000073c067.ZIP](https://github.com/Atmosphere-NX/Atmosphere/files/10015349/report_000000000073c067.ZIP)

SciresM, 2022-12-04:
1) You did not use the issue templates,

2) This report indicates a hardware issue with your RAM or otherwise. Nothing to be done about it in software.

closed 2022-12-04 by SciresM

genjiow2, 2022-12-04:
Is this the emmc module? If I order a new one, will the error go away?

SciresM, 2022-12-04:
No. Previously this has been the actual physical RAM, though it could be the CPU.

To verify, what happens when you try to boot stock-without-atmosphere on this device?

genjiow2, 2022-12-04:
> Нет. Раньше это была фактическая физическая оперативная память, хотя это мог быть и ЦП.
> 
> Чтобы проверить, что происходит, когда вы пытаетесь загрузить стоковую без атмосферы на этом устройстве?

I can run OFW and hekate,but error only in atmosphere

SciresM, 2022-12-04:
"run hekate" <-- unrelated.

Well, the error report you uploaded results abort in `ams::kern::`anonymous namespace'::LoadInitialProcessBinaryHeader`. This occurs due to malformed INI1 binary. I would guess the bad RAM is in the later parts; the INI is copied to end of RAM, and atmosphere's INI is larger than stock's.

Either way, issue/recommendation is the same.

SciresM, 2022-12-04:
it's a hardware problem, so I am unable to speculate.

genjiow2, 2022-12-05:
> это аппаратная проблема, поэтому я не могу предположить.

Could this be due to the fact that the set-top box has an sxcore chip or something similar?

genjiow2, 2022-12-08:
> Я исправил это, просто заменив карту micro sd, та, которую я использовал, была неисправна.

Really?I tried 3 sd-cards,but same error...

genjiow2, 2022-12-09:
> Нет. Раньше это была фактическая физическая оперативная память, хотя это мог быть и ЦП.
> 
> Чтобы проверить, что происходит, когда вы пытаетесь загрузить стоковую без атмосферы на этом устройстве?

Unfortunately, you were right. The problem is with the CPU, it is in a very terrible state. For future readers: Never buy used consoles

Masamune3210, 2022-12-09:
Well, not never, just always remember there is a chance that there is prior damage and that you don't know the history of the usage and behavior, just like anything else

genjiow2, 2022-12-18:
So, after a couple of weeks, the problem was solved by reballing. To future readers with the same problem: take the console to a service or change the SD card