Atmosphere-NX/Atmosphere #1995: 524850534f53454d error Trying to Load CFW Games

kinseta, 2022-11-18:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?
I recently just downloaded and installed some CFW games. Every-time I try to start them I get a Panic Error, and have to reboot. Everything else on Works Fine, Tinfoil, HB Menu, Retroarchthe only problem I’m having is with starting the Games I downloaded from the shop. (Yes I have sig patches etc)
[ Describe the issue in detail and what you were doing beforehand. ]</br> Tried to play a game 
[ Did you make any changes related to Atmosphère itself? ]</br> no
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

masagrator, 2022-11-18:
Piracy is not supported.

kinseta, 2022-11-18:
Fixed

Masamune3210, 2022-11-18:
Not even slightly. Piracy is still not supported.

kinseta, 2022-11-19:
Nah problem on my end is fixed. Lol move round

kinseta, 2022-11-19:
> Not even slightly. Piracy is still not supported.

It’s my Game goofy

closed 2022-11-19 by kinseta