Atmosphere-NX/Atmosphere #1996: Bug report

gcasasf, 2022-11-19:
## Bug Report

### What's the issue you encountered?

I recently updated my switch firmware to 15.0.1 and then my Atmosphère to 1.4.0 and hekate to 5.9.0

Whenever I try to run syscfw from hekate or on its own I get a fatal error. This happens every time and I can not get around it.

It will say the following

A fatal error occurred when running Atmosphere.
Title ID: 010000000000002b
Error Desc: std : : abort ( ) called (0xffe)
Report saved to /atmosphere/fatal_errors/report_00000000092d12a8.bin
Press POWER to reboot.

### How can the issue be reproduced?

This issue occurred when I tried to update and it happens every time I attempt to run syscfw

### Crash Report

[report_000000001dac8923.zip](https://github.com/Atmosphere-NX/Atmosphere/files/10047788/report_000000001dac8923.zip)

### System Firmware Version

15.0.1

### Environment?

What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Both fusee and hekate
Fusee is the one included in the zip for the latest atmosphere update and hekate is 5.9.0 

### Additional context?

Serial number XAJ40013587297 bought in 2017

masagrator, 2022-11-19:
> Report saved to /atmosphere/fatal_errors/report_00000000092d12a8.bin

uploaded crash report doesn't match mentioned file name, can you provide the one mentioned there?

gcasasf, 2022-11-19:
sorry the mentioned is this one report_000000001dac8923.bin

Masamune3210, 2022-11-19:
That does not match the fatal error though, please upload the one mentioned in the error report

gcasasf, 2022-11-19:
[report_00000000222abecc.zip](https://github.com/Atmosphere-NX/Atmosphere/files/10047890/report_00000000222abecc.zip)

Got this trying to boot now from the same kind of error.

Thank you

gcasasf, 2022-11-21:
What should I do then? How can I know it?

SciresM, 2022-12-04:
Duplicate of #1999.

You have a corrupt PRODINFO/system save.

Follow the instructions/directions in https://github.com/Atmosphere-NX/Atmosphere/issues/1256 for guidance. You'll need to restore a non-corrupt backup and/or delete the system settings savedata.

closed 2022-12-04 by SciresM