Atmosphere-NX/Atmosphere #1999: Fatal error

gcasasf, 2022-11-27:
### What's the issue you encountered?

I recently updated my switch firmware to 15.0.1 and then my Atmosphère to 1.4.0 and hekate to 5.9.0

Whenever I try to run syscfw from hekate or on its own I get a fatal error. This happens every time and I can not get around it.

It will say the following

A fatal error occurred when running Atmosphere.
Title ID: 010000000000002b
Error Desc: std : : abort ( ) called (0xffe)
Report saved to /atmosphere/fatal_errors/report_00000000222abecc.bin
Press POWER to reboot.

### How can the issue be reproduced?
This issue occurred when I tried to update and it happens every time I attempt to run syscfw

### Crash Report

[report_00000000222abecc.zip](https://github.com/Atmosphere-NX/Atmosphere/files/10098053/report_00000000222abecc.zip)

### System Firmware Version

15.0.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:

What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Both fusee and hekate
Fusee is the one included in the zip for the latest atmosphere update and hekate is 5.9.0

### Additional context?
Serial number XAJ40013587297 bought in 2017

skalidindi, 2022-11-28:
I am facing this same issue. Same setup.

AzleyITA, 2022-12-04:
i have the same issue :/

SciresM, 2022-12-04:
Ah, sorry, wrong issue.

SciresM, 2022-12-04:
Oh, no, wait, it was correct.

You have a corrupt PRODINFO/system save.

Follow the instructions/directions in https://github.com/Atmosphere-NX/Atmosphere/issues/1256 for guidance. You'll need to restore a non-corrupt backup and/or delete the system settings savedata.