Atmosphere-NX/Atmosphere #2002: Atmosphere 1.4.0 "Game Card Could Not be Read" Error on latest 15.0.1 firmware update.

OgSilentcell, 2022-11-30:
I started this with a completely new 4.0 whatever Switch. Did the jailbreak per many YT Videos everything went well. Wanted to gain full RAM access so I bought Bayonetta. Inserted it and it said I needed an update to play it. Went on the YT's again to see how to update. Got the AIO Updater tool and now I CFW won't read the card so as you would imagine w/o full RAM access I can't run things like Retroarch without having problems like "missing assets" that I can't update because that keeps giving me an error. It isn't a card reader issue because when I do want to play Bay I just load the stock firmware and everything is fine. I tried to do the "force-disable nogc" fix, by editing the "stratosphere.ini" but I got nothing with that so any thoughts on what might be going on? Thanks!

masagrator, 2022-11-30:
First of all use issue template. And next upload your stratosphere.ini

OgSilentcell, 2022-12-04:
ssue: Atmosphere 1.4.0 Game Card Could Not Be Read On Latest Firmware 15.0.1

Describe the bug
After using AIO Updater to update Nintendo Switch Firmware from 4.0 to 15.0.1 running Atmosphere 1.4.0 My Switch can no longer recognize game cards that are inserted and is instead met with the "game card could not be read. Please reinsert."

To Reproduce
Steps to reproduce the behavior:
*Note I'm not pushing this channel in anyway shape or form and I'm not even saying it was the cause, this is just where I got my initial information from

1. Go to (https://www.youtube.com/watch?v=ardN1T0CuvI)
2. Use this guide to jailbreak using RCM Jig
3. Once you get to the 22:25 skip the Emunand partition section
4. Skip to about 34:19 get the new package this dood makes from discord
5. Plug your switch back in and use USB Tools to transfer and over write the existing files on the SD card
6. Reboot and try to launch Atmosphere.
7. An Error will pop up saying "Forced Emunand" or something like that.
8. You'll then be taken back to the Hekate Homescreen
9. Go back into USB Tools and connect back to your computer.
10. Delete the Atmosphere folder that's currently on the SD card and replace it with a fresh download straight from Github.
11. Eject your Switch, Reboot, you should now be able to launch back into custom Firmware
12. Insert Bayonetta
13. A notification to update Switch Firmware will pop up.
14. Go to the HomeBrew shop and download AIO Updater
15. download and install the latest firmware 15.0.1 and install
16. Boot back into Atmosphere

Expected behavior
You should now come across the Error that the game card could not be read.

Screenshots
If applicable, add screenshots to help explain your problem.

Desktop (please complete the following information):

OS: [e.g. iOS]
Browser Brave
Version 1.45.133
Smartphone (please complete the following information):

Device: Note 9
OS: Android 10
Browser Brave
Version 1.45.133

Additional context
So like I said I'm not shilling for this channel and I'm not saying it was the cause. I realize I'm a bit of a mess, but this is everything I can remember about how I ended up where I did.

Optional additional items

K:\atmosphere\config_templates

[stratosphere]
; To force-enable nogc, add nogc = 1
; To force-disable nogc, add nogc = 0
nogc = 0

Issue default title:
Assignees:
Labels:

SciresM, 2022-12-04:
Right, this doesn't really use the issue templates properly, also this is just a nogc problem.

"get the new package this dood makes from discord" <-- support for unofficial packages not provided.

Also, this issue tracker is for bugs, not for support, please seek support on discord.

closed 2022-12-04 by SciresM

OgSilentcell, 2022-12-04:
I'm sorry, I'm really trying here. I'm not trying to be a pain and at the same time I'm not trying to insult anyone's intelligence around here that's the reason for the "short hand." I guess when I said "this doods package" I just meant that,  that was a step I took that lead me to where I'm at. I suppose it's just for context more that anything else. I'm not trying to use what this guy put together because even when I did everything on a new SD with the bare minimum files I got the same message.  C'est la vie I suppose..... I'll try discord like you said and see if they know how to help. Thanks for the quick replies.

SciresM, 2022-12-04:
I don't think you're being a pain or that you're trying to insult anyone's intelligence. I understood that you meant that was a step you took.

Regardless, though, this is going to be a support issue ^^

OgSilentcell, 2022-12-04:
I understand now. Thank you for the clarification. I'll see if I can get this right the next time. :)

2play, 2023-01-11:
> Also, if you booted through hekate, make sure no-gc is disabled there as well

By simply disabling the AutoGC from ON to OFF the original cards start ok with no error.
(Make sure your sigpatches are updated or use AWOO installer to update them.
Thank you all for the GC clarification