Atmosphere-NX/Atmosphere #2007: Crash Report

MikeyProHD, 2022-12-18:
I was playing Smash Ultimate when all of a sudden it crashed. So I closed out of it and I pressed the power button to but it into sleep mode. When I woke the system back up and opened another game it crashed.

MikeyProHD, 2022-12-18:
Atmosphère Crash Report (v1.7):
Result:                          0x4A8 (2168-0002)

Process Info:
    Process Name:                Application
    Program ID:                  01006a800016e000
    Process ID:                  000000000000009b
    Process Flags:               00000077
    User Exception Address:      000000648f1b6000
Exception Info:
    Type:                        Data Abort
    Address:                     000000000000000c
    Fault Address:               0000000000000000
Crashed Thread Info:
    Thread ID:                   000000000000489f
    Thread Name:                 MainThread
    Stack Region:                0000006989e15000-0000006989f15000
    Registers:
        X[00]:                   0000000000000000
        X[01]:                   00000013c26bf5d3
        X[02]:                   00000000012003a8
        X[03]:                   0000006989f11690
        X[04]:                   0000006989f11680
        X[05]:                   0000006989f11670
        X[06]:                   0000000000000000
        X[07]:                   00000000ffffffff
        X[08]:                   0000000000000000
        X[09]:                   0000006e9b4fc2a0
        X[10]:                   0000000000000018
        X[11]:                   000000297ef7a000
        X[12]:                   000000000d000000
        X[13]:                   000000004c3b4084
        X[14]:                   0000000000000000
        X[15]:                   0000006eabfdc978
        X[16]:                   0000002d98802b08
        X[17]:                   000000297e5678d8 (Skyline + 0x88d8)
        X[18]:                   0000000000000000
        X[19]:                   00000013c26bf5d3
        X[20]:                   0000000000ffffff
        X[21]:                   0000000000000000
        X[22]:                   0000002d98802000
        X[23]:                   000000297bd36000
        X[24]:                   0000006e58dd9150
        X[25]:                   00000013c26bf5d3
        X[26]:                   0000006ed213e958
        X[27]:                   0000000050000000
        X[28]:                   0000000000000000
        FP:                      0000006989f116f0
        LR:                      0000002d987d20a8 (one-slot-eff + 0x20a8)
        SP:                      0000006989f114b0
        PC:                      0000002d987d20ac (one-slot-eff + 0x20ac)
    Stack Trace:
        ReturnAddress[00]:       0000002976e59a00 (cross2_Release.nss + 0x455a00)
        ReturnAddress[01]:       0000002976de367c (cross2_Release.nss + 0x3df67c)
        ReturnAddress[02]:       0000002976dac93c (cross2_Release.nss + 0x3a893c)
        ReturnAddress[03]:       0000002979da7aa4 (cross2_Release.nss + 0x33a3aa4)
        ReturnAddress[04]:       0000002976db1774 (cross2_Release.nss + 0x3ad774)
        ReturnAddress[05]:       0000002977d7f588 (cross2_Release.nss + 0x137b588)
        ReturnAddress[06]:       0000002977d60bfc (cross2_Release.nss + 0x135cbfc)
        ReturnAddress[07]:       0000002977d6374c (cross2_Release.nss + 0x135f74c)
        ReturnAddress[08]:       0000002978254484 (cross2_Release.nss + 0x1850484)
        ReturnAddress[09]:       0000002978254378 (cross2_Release.nss + 0x1850378)
        ReturnAddress[10]:       000000297a127d14 (cross2_Release.nss + 0x3723d14)
        ReturnAddress[11]:       0000002978ce2570 (cross2_Release.nss + 0x22de570)
        ReturnAddress[12]:       000000297a127d14 (cross2_Release.nss + 0x3723d14)
        ReturnAddress[13]:       0000002978ce2570 (cross2_Release.nss + 0x22de570)
        ReturnAddress[14]:       000000297a127d14 (cross2_Release.nss + 0x3723d14)
        ReturnAddress[15]:       000000297a14f5a8 (cross2_Release.nss + 0x374b5a8)
        ReturnAddress[16]:       0000002976cc9cf0 (cross2_Release.nss + 0x2c5cf0)
        ReturnAddress[17]:       000000297e822a2c (nnSdk + 0x29da2c) (_ZN2nn4init5StartEmmPFvvES2_ + 0x4c)
[... 2583 lines cut ...]
                                 0069b657aeb0 a8 53 4f 59 6e 00 00 00 00 00 00 00 00 00 00 00
                                 0069b657aec0 38 53 4f 59 6e 00 00 00 e8 52 4f 59 6e 00 00 00
                                 0069b657aed0 d8 52 4f 59 6e 00 00 00 ec 52 4f 59 6e 00 00 00
                                 0069b657aee0 78 53 4f 59 6e 00 00 00 98 53 4f 59 6e 00 00 00
                                 0069b657aef0 e0 52 4f 59 6e 00 00 00 00 00 00 00 c0 03 00 00
                                 0069b657af00 a8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                 0069b657af10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    TLS Address:                 00000064298f4400
    TLS Dump:                                 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
                                 0064298f4400 00 00 00 00 0c 00 00 00 00 00 00 00 00 00 00 00
                                 0064298f4410 53 46 43 4f 00 00 00 00 00 00 00 00 01 97 b0 7a
                                 0064298f4420 a8 00 00 00 c0 03 00 00 ad 03 00 00 00 00 00 00
                                 0064298f4430 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                 0064298f4440 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                 0064298f4450 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                 0064298f4460 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
                                 0064298f44a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                 0064298f44b0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                 0064298f44c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                 0064298f44d0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                 0064298f44e0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                 0064298f44f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Masamune3210, 2022-12-18:
Please go back, remove the pasted crash log, attach it like you are supposed to and follow the rest of the template while you are at it. The information is required for a reason, not for you to just blow it away and post whatever you want

MikeyProHD, 2022-12-18:
Didn't see the thread at first. Here you go: https://github.com/Atmosphere-NX/Atmosphere/issues/2008

MikeyProHD, 2022-12-18:
(https://github.com/Atmosphere-NX/Atmosphere/issues/2008)

MikeyProHD, 2022-12-18:
Sorry, that thread didn't show up when I did it. Should be fixed now.

closed 2022-12-18 by SciresM