Atmosphere-NX/Atmosphere #2008: Random Crash

MikeyProHD, 2022-12-18:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

I was playing Smash Ultimate when all of a sudden it crashed. So I closed out of it and I pressed the power button to but it into sleep mode. When I woke the system back up and opened another game it crashed.

### How can the issue be reproduced?

Probably by making Smash Ultimate crash, exiting out, going into sleep mode and launching another game.

### Crash Report

https://gist.github.com/MikeyProHD/b71f4e2cb9fba95e40af673a02fb8062

### System Firmware Version

15.0.1
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Fusee
- Official release or unofficial build: 
  - [ Official release version N/A

- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: N/A
- EmuMMC or SysNAND: Not sure
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

Masamune3210, 2022-12-18:
Go back and actually fill out the issue template

SciresM, 2022-12-18:
You have skyline/mods installed. I have zero reason to believe this is an atmosphere issue.

closed 2022-12-18 by SciresM

MikeyProHD, 2022-12-18:
I figured it out. That was why actually. I had unnecessary skyline stuff that caused it.