Atmosphere-NX/Atmosphere #2009: Crash leading to corrupted emuMMC; Unable to boot Atmosphere error 0x207202 (2002-4153)

HeyItsJono, 2022-12-23:
## Bug Report

### What's the issue you encountered?

I'm just filing this report on behalf of my younger brother. He was playing a game and then exited to the home menu with the home button, the system then froze and atmosphere crashed. I unfortunately did not get the initial error he experienced as he then attempted to restart. Any subsequent attempts to boot back into atmosphere via RCM lead to a crash screen singling out the home menu's title ID as the cause of the crash.

Atmosphere would not boot both by directly attempting to inject the latest fusee.bin, or trying to launch via Hekate (both package3 and fusee.bin methods didn't work). Hekate booted and functioned without issue, as did TegraExplorer and Lockpick_RCM.

The issue persisted after deleting the atmosphere and bootloader folders and doing a 'clean install' by redownloading from the latest Github releases. I saved the crash logs/dumps (attached) before reinstalling.

At all times the microSD used was a 512GB legitimate Samsung card that I had personally tested (with a rather lengthy h2testw run) when we first bought it a few months ago. It has always been formatted as FAT32.

The EmuMMC partition was able to be mounted in NxNandManager via Hekate USB mounting, and in this way I was able to do a NAND backup of this unbootable EmuMMC and in particular was able to save the USER partition and dump the saves from it to be restored later. I was then able to restore an old backup of his EmuMMC using NxNandManager and we are now able to boot atmosphere normally again.

[I found this to be a handy resource for anyone in a similar position who needs to backup saves.
](https://suchmememanyskill.github.io/guides/switchextractsaves/)

I'm posting the report with the crash logs/dumps in case there's anything here that could help get to the source of why this happened in the first place, seems unfortunate that an atmosphere crash might potentially corrupt the EmuMMC.

### How can the issue be reproduced?

I don't believe it could be reproduced reliably.

### Crash Report
[01671624059_0100000000001000.log](https://github.com/Atmosphere-NX/Atmosphere/files/10294581/01671624059_0100000000001000.log)

[01671623863_0100000000001000.log](https://github.com/Atmosphere-NX/Atmosphere/files/10294582/01671623863_0100000000001000.log)

[01671623406_0100000000001000.log](https://github.com/Atmosphere-NX/Atmosphere/files/10294587/01671623406_0100000000001000.log)

[bin files.zip](https://github.com/Atmosphere-NX/Atmosphere/files/10294597/bin.files.zip)

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate, but also unable to boot directly via fusee.bin injection.
- Official release or unofficial build: Official Release 1.4.0-master-36a33617a
- Do you have additional kips or sysmodules you're loading: sigpatches (to load backups of carts we own, no piracy, as well as a hbl nsp), ldn_mitm; I don't believe he had other sysmodules enabled at the time.
- Homebrew software installed: None running at the time except ldn_mitm which was enabled in the background as mentioned
- EmuMMC or SysNAND: Partition-based EmuMMC

SciresM, 2022-12-23:
`fs::ResultUnclearedRomRealDataVerificationFailed` -- this is just corruption. It's not really a bug, and not really due to anything atmosphere's doing. If you want to make this an issue, it's probably better off on the emummc repo.

Also:
> sigpatches (to load backups of carts we own, no piracy)

This is piracy, and piracy is not supported.

closed 2022-12-23 by SciresM

HeyItsJono, 2022-12-23:
Thanks for taking a look and confirming it was a corrupted EmuMMC, I was suspicious that was the case but it's nice to have it confirmed. Does this kind of thing just happen by chance or is there anything else I can do to minimize the risk of just randomly ending up with a corrupted EmuMMC?