Atmosphere-NX/Atmosphere #2012: Incorrect Fusee Version (0xb46dcf71)

TheOneAmberbug, 2023-01-05:
## Bug Report

### What's the issue you encountered?

I wanted to play a game with mods so installed mods in atmosphere. Didn't work. Wiped the SD and only installed the master still the same error.

### How can the issue be reproduced?

[ * ]</br>
I installed the master and downloaded the fusse.
Then injected and it comes.

### Crash Report

No existing

### System Firmware Version

I'm on firmware 15.0.1 

### Environment?

Using only atmosphere (the master file with everything) and TegraRCMGui with the fusee.bin from the same release.

### Additional context?

It isn't my first time jailbreaking

Masamune3210, 2023-01-06:
Either you are accidentally pushing the wrong file or you have missed something. If something this simple was failing en masse, the issues would be full of reports by now

goofwear, 2023-01-06:
I updated a week or so ago and had the same problem. I had to use an older fusee

TheOneAmberbug, 2023-01-07:
Someone gave me the tip to use an older fusee.bin and now it says Invalid GPT(or GPI) Signature. Could it be related to EmmuMC that I had in the past? When yes where should I ask for help?

goofwear, 2023-01-10:
Sorry I had to removed my post yesterday after I realized I was on the wrong thread. I thought you were talking about the fuse then from the Hekate. I updated the NX stuff. And then on my loader I had to get the slightly older hekate fuse to get mine to boot. Sorry for any confusion.

SciresM, 2023-01-18:
This is always user error, you need to make your payload and what's on SD be the same.