Atmosphere-NX/Atmosphere #2013: Atmosphere suddenly crashing

R-son, 2023-01-07:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[image]
I get this type of errors when trying to launch a software (gallery or game) under atmosphere. When I don't launch anything for some time something else happens
[image]
I didn't actually do anything. Like I didn't download anything since the last atmosphere update in October 2022. The bug just started happening a few days ago. It also corrupts my games' data when I try to run them on atmosphere

### How can the issue be reproduced?

N/A

### Crash Report

https://gist.github.com/R-son/01b3d0eb12373c89eafb8f71a02326a1

### System Firmware Version

15.0.1

### Environment?

I use hekate as a bootloader because it seemed to be crashing more often with fusee
I used official releases, 6.0.1 for hekate and the fusee version that is coupled with the 1.4.0 version of atmosphere

### Additional context?

As suggested, I tried to delete the contents folder, to fix the archive bits through hekate and to run sdcard through h2testw.
None of those actions fixed or detected the problem.
I then reinstalled the whole thing on another sd card and it still didn't work

Hielkenator, 2023-02-15:
I also have this exact problem , also I get a crash and sd card corruption when trying to delete user accounts. Switch gets extremely slow after update fw 15.01 icw atmosphere 14.0 and 14.01

1332348216, 2023-03-06:
Last question, maybe you can format the memory card as FAT32

R-son, 2023-03-07:
> Last question, maybe you can format the memory card as FAT32

It was already in this format, I formatted it again to make sure before I posted this issue. It still doesn't work to this day and idk why

Grip555, 2023-03-27:
I have the exact same issue.

mnsg23, 2023-04-27:
I've recently been dealing with an identical problem when trying to load Super Smash Bros. Ultimate, with my system version being 16.0.2, while the Atmosphere version is 1.5.2. It read...

_Panic occurred while running Atmosphere.

Title ID: 010041544D530000
Error: std::abort (0xFFE)

Report saved to /atmosphere/fatal_errors/report_0a40b32d9.bin_

And after a lot of troubleshooting, I tried moving the software data from my microSD card, onto the system memory. So far it's working, but I'm still keeping tabs on the matter.

Sensei305, 2023-06-15:
Having the same issues. Total noob at this tho I thought I f'ed it up somehow. Been in APmode since I installed, no wifi connect, literally didn't change or make any new installs or updates. Just played Tears. A couple days ago game started crashing giving me software errors like the errors mentioned above out of nowhere.

closed 2025-04-03 by SciresM