Atmosphere-NX/Atmosphere #2015: Sky: Children of the Light Crashes

TurtleP, 2023-01-16:
## Bug Report

### What's the issue you encountered?
I've been playing the game for some time now today and it's crashed twice.

### How can the issue be reproduced?
I don't think it's easily reproducible, but: first was when skipping a cutscene in-game when joining a friend and the other when collecting a spirit.

### Crash Report

[01673902526_0100c52011460000.log](https://github.com/Atmosphere-NX/Atmosphere/files/10429430/01673902526_0100c52011460000.log)

[01673900103_0100c52011460000.log](https://github.com/Atmosphere-NX/Atmosphere/files/10429431/01673900103_0100c52011460000.log)

### System Firmware Version
15.0.1

### Environment?
- I am launching official atmosphere 1.4.0 with Hekate.
- No additional KIPs or sysmodules loaded
- No homebrew software "installed".
  - The only apps I have downloaded to execute on hbmenu via title takeover are:
    - ftpd, Checkpoint, Reboot to Payload and LÖVE Potion
- EmuMMC or SysNAND: **SysNAND**

### Additional context?
The game itself is up-to-date (0.20.0.209642 from its details on the HOME Menu).

SciresM, 2023-01-17:
Do you know if the game crashes occur when playing without atmosphere? One of those crashes is a data abort, lol.

TurtleP, 2023-01-17:
I have no idea, honestly. I could try running it stock and play some tomorrow, if I get a chance. What's the other crash?

SciresM, 2023-01-17:
a user break of some kind. I don't actually own the game, so I have no idea.

alula, 2023-02-23:
(it's free on eShop)

NWPlayer123, 2023-02-23:
I've played 110+ hours of this game on the same-ish Atmosphere setup (I have sys-clk on but not changed from stock for this title) and it's been pretty stable aside from game jank so I'm inclined to say it's that over something with Atmosphere

JeDaYoshi, 2023-08-23:
Sky is known to be kinda prone to crashes, not only on Switch but on every platform it's available for.
Like Alula mentions, it's free on the eShop for further investigation/testing, but I'd place the bets on the game to be the one on fault rather than Atmosphère.

@TurtleP Are you still experiencing crashes?

closed 2025-04-03 by SciresM