Atmosphere-NX/Atmosphere #2019: Switch won't boot into Atmosphere from Hekate

SavageColeTrain, 2023-01-18:
## Bug Report

### What's the issue you encountered?

[ tried to boot into Atmosphere from Hekate, gives a yellow error that says Panic occured while running Atmosphere and a title ID 0100000000000000 error UNK <0xE>]</br>
[ none that I am aware of ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ boot into Atmosphere from Hekate, gives a yellow error that says Panic occured while running Atmosphere and a title ID 0100000000000000 error UNK <0xE>. ]</br>

### Crash Report

### System Firmware Version

15.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: 
  - [ Official release version 14.0.1 (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading: sys-clk, Mission Control, ldl_mitm
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND: EmuMMC, File based
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2023-01-18:
Upload the report.

SavageColeTrain, 2023-01-18:
I wasn't able to for some reason. Does it matter if it's a .bin file?

SciresM, 2023-01-18:
Yes. You need to upload the bin file.

SavageColeTrain, 2023-01-18:
Can I just send it from here?

SciresM, 2023-01-18:
I don't care how you send it, but I cannot help you without the report file.

SciresM, 2023-01-18:
I will re-open if the issue is real, when report is uploaded

closed 2023-01-18 by SciresM