Atmosphere-NX/Atmosphere #2020: Crash on Unravel Two

CarlosSLoureiro, 2023-01-19:
## Bug Report

### What's the issue you encountered?
When I try to access the Chapter II in the game Unravel Two the game crashes.
I'm about sure that it's not new. I've noticed this problem for some time now but just reported right now...

### How can the issue be reproduced?
Just try to enter the Chapter II in the game Unravel Two

### Crash Report

[01674159053_0100e5d00cc0c000.zip](https://github.com/Atmosphere-NX/Atmosphere/files/10460725/01674159053_0100e5d00cc0c000.zip)

### System Firmware Version

15.0.1

### Environment?
- Atmisphère 1.4.0 with Hekate 6.0.1
- Only Tesla enabled at sysmodules
- I'm also using EmuMMC

### Additional context?

- No additional context

SciresM, 2023-01-19:
Does the error occur when you have no custom sysmodules? Please test, and post a screenshot showing nothing inside /atmosphere/contents to verify.

CarlosSLoureiro, 2023-01-19:
Thank you for the suggestion. The crash its because of the custom translation added to the contents directory. I didn't even remember that..

closed 2023-01-19 by CarlosSLoureiro