Atmosphere-NX/Atmosphere #2021: Modding romfs in Fire Emblem: Engage causes Atmosphere to crash.

ayssia, 2023-01-20:
## Bug Report

### What's the issue you encountered?

I've unpacked FE:Engage and modded a file inside it.
I placed the modded file under `atmosphere\contents\TITLEID\romfs\Data\...` like what I did to other games, and atmosphere crashed.
Further testing showed that even without modding the file it would cause a crash.

### How can the issue be reproduced?

1. Unpack Fire Emblem: Engage.
2. Choose a file at random. For example, `romfs\Data\StreamingAssets\aa\Switch\fe_assets_gamedata\person.xml.bundle`. No need to modify this file.
3. Copy the unchanged file to `atmosphere\contents\` .
4. Run the game. Atmosphere crashed.

`A fatal error occured when running Atmosphere.`
`Error Desc: Data Abort (0x101)`

### Crash Report

It's a binary file so I converted it to base64: https://gist.github.com/ayssia/fd0f4901740362030c8587e46e89e393

### System Firmware Version

15.0.1

### Environment?

Hekate 5.9.0 with Official Atmosphere 1.4.0, SysNAND. Test with all kips/sysmodules disabled.

### Additional context?

I guess the crash is caused by too many files in the unpacked game, there are ~178,000 files.

SciresM, 2023-01-20:
Yeah. It's an out of memory.

The trivial fix is to add 12 MB of memory to ams.mitm. I'm thinking about whether there's some strategy to be used that reduces memory better than that...it's just way too many files.

GuhLucena, 2023-01-24:
Is there any way to increase the memory of ams.mitm in some configuration file, or do we still have to wait for a solution?

SciresM, 2023-01-24:
You could of course build yourself in the meantime. Just make this 28 MB (lol): https://github.com/Atmosphere-NX/Atmosphere/blob/master/stratosphere/ams_mitm/source/amsmitm_main.cpp#L27

BeckysFootSlave, 2023-01-24:
> You could of course build yourself in the meantime. Just make this 28 MB (lol): https://github.com/Atmosphere-NX/Atmosphere/blob/master/stratosphere/ams_mitm/source/amsmitm_main.cpp#L27

Just a dumb question!
Is 28 the highest or is it possible to use 32?
Just asking because of that 2,4,8,16,32,... thing!

commit 0dd071b2 (2023-01-24): romfs: thank you fire emblem, very cool (closes #2021)
closed 2023-01-24 by SciresM