Atmosphere-NX/Atmosphere #2031: Atmosphère Fatal Error "Data Abort (0x101)"

wockyslushDOTjpg, 2023-02-10:
Bug Report [Fatal Error Occurred when running Atmosphére.]

### What's the issue you encountered?

So, my copy of Fire Emblem Engage works perfectly fine otherwise, but when I use "nxdumptool" on it
and attempt to open the game I'm left on a black screen and get this error message,

 " A fatal error occurred while running Atmosphere.
 Program ID: 010041544d530000, Error Desc: Data Abort (0x101)"

### **Describe the issue in detail and what you were doing beforehand.** 

The game was working fine before like I said, and then I decided to use nxdumptool to extract the game's files, though it seems like no one else in the community is having the same issue as me. I wrote a potential cause as to why this is happening to me in the "Context" section of this post for reference. 

### Did you make any changes related to Atmosphère itself?

No.

### How can the issue be reproduced?

1.   With a fresh SD Card, Install Fire Emblem Engage to your system; it should play like normal.
2.   Then spend like 3 hours dumping the game files through nxdumptool and start the game.
3.   If it works for you proceed to berate me.

### Crash Report

Atmosphère Fatal Error:
https://gist.github.com/wockyslushDOTjpg/70850539910aa4f06e64720d905bc156

Atmosphère Crash:
https://gist.github.com/wockyslushDOTjpg/9a20000a11896fbedc6b557e256024d1

Not sure how related the crash is to the fatal error, but better safe than sorry.

### System Firmware Version

15.0.1

### Environment?

Hekate launched by TegraRcmGUI :

 Official release version "v6.0.1" (hekate_ctcaer_6.0.1.bin)

 No additional kips or sysmoduels 

 Homebrew software installed: v3.5.1

 SysNAND

### Additional context?

Pretty Important:  

I'm worried this may have to do with my SD Card. I used to have an issue with my SD Card initializing in 1-bit mode, so I replaced the reader and the problem, seemingly, went away; It could be that I just haven't activated the "initialized in 1-bit mode" error again yet, so the error may be coming from my ongoing Hardware issue.

I truly apologize if this answer is super obvious to anyone here, I don't mean to be annoying, but if the issue is unrelated to my Hardware then I'd rather just ignore the SD Card part until I can afford to fix it permanently. Also, I'm just now realizing nothing else is wrong with my system except this one thing I've been avoiding.

SciresM, 2023-02-10:
"dumping the game files through nxdumptool and start the game" why not just...dump as an NCA and extract on PC using hactool...?

Anyway, it's not even possible to dump the game files on console to SD, FE:E has more files within a single directory than FAT supports, and this can/will cause OS instability. 

The crash you're sharing is in the home menu, not ams code, so I'm going to close this. I'm not really sure why you're trying to do whatever you're trying to do.

SciresM, 2023-02-10:
Actually, I think you uploaded completely unrelated files. You probably meant to upload a binary file from inside the fatal_errors folder.

closed 2023-02-10 by SciresM

wockyslushDOTjpg, 2023-02-10:
Oh, alright thanks! I actually had thought about dumping the files to my PC after the 2nd time this happened, but I guess I was being a bit stubborn, and yes; I did mean to upload the binary file from fatal_errors. 

Thanks again.

SciresM, 2023-02-10:
Yeah. You should upload the binary file so that I can verify the issue is what I'd expect (too many files --> oom), but yeah.

wockyslushDOTjpg, 2023-02-10:
Gotcha. Not sure how these are usually uploaded but I couldn't just drag and drop, unfortunately.

[report_000000004840c723.zip](https://github.com/Atmosphere-NX/Atmosphere/files/10711402/report_000000004840c723.zip)

SciresM, 2023-02-10:
Yeah, this is exactly what I'd expect. Basically, nxdumptool is opening the romfs from another process, and this will cause the romfs to be built for mod support...this uses almost all available memory. And then you're launching the game before all that memory is freed (takes a little while) --> it's built again, and shit gets exhausted.

This makes sense, it's not really fixable, best I can do is "don't dump the files".

Thank you for uploading! You did exactly the correct thing by zipping it :)

DarkMatterCore, 2023-02-11:
Well, nxdumptool parses the underlying NCA FS sections on its own by performing raw NCA reads via ncm. It doesn't really retrieve a FS handle through another process -- and it's been like this for years now.

If I had to guess...

> Anyway, it's not even possible to dump the game files on console to SD, FE:E has more files within a single directory than FAT supports, and this can/will cause OS instability.

I'd say this is the real issue here. That's just not being handled at all in nxdumptool.