Atmosphere-NX/Atmosphere #2032: Tried to update atmosphere but now I get "Panic occured. Title ID : 010041544D53000"

Tendiemancome, 2023-02-11:
I tried to update Atmosphere manually on my Mac and now when I inject fusee directly and also when I try to boot into atmosphere through Hekate I get 
"Panic occured. Title ID : 010041544D53000"

Things I have tried (everything on the forums I could find);
Clean install of Atmosphere and bootloader folder.
Replaced copy of prodinfo. with my backup copy.
Set archive bit through hekate and also tried manually with terminal on my Mac.
My SD card is formatted to FAT32
Deleted contents of "Contents" folder.
Always use "CleanEject" when dismounting the SD from the Mac.

### Crash Report

### System Firmware Version
atmosphere-1.4.1
OFW 15 0 0
Hekate 6.0
SysNAND

Side note, am trying to load my report.bin file with gist github, drag and drop wont work :(

SciresM, 2023-02-11:
You can't he helped without the report file. Zip it and upload it and if the issue is real I will reopen.

closed 2023-02-11 by SciresM

Tendiemancome, 2023-02-12:
[report_05782431.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/10715352/report_05782431.bin.zip)

Thanks for highlighting the.zip upload option