Atmosphere-NX/Atmosphere #2033: Tried to update atmosphere but now I get "Panic occured. Title ID : 010041544D53000"

Tendiemancome, 2023-02-12:
I tried to update Atmosphere manually on my Mac and now when I inject fusee directly and also when I try to boot into atmosphere through Hekate I get
"Panic occured. Title ID : 010041544D53000"

Things I have tried (everything on the forums I could find);
Clean install of Atmosphere and bootloader folder.
Replaced copy of prodinfo. with my backup copy.
Set archive bit through hekate and also tried manually with terminal on my Mac.
My SD card is formatted to FAT32
Deleted contents of "Contents" folder.
Always use "CleanEject" when dismounting the SD from the Mac.

System Firmware Version
Atmosphere 1.4.1
OFW 15 0 0
Hekate 6.0.1
SysNAND
[report_0546d337.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/10717091/report_0546d337.bin.zip)

(Thanks for highlighting .zip will upload)

masagrator, 2023-02-12:
When you use mass storage on Mac, you must run archive bit fixer EVERY TIME you eject device. You were doing this only once or all times?

Tendiemancome, 2023-02-12:
Hi Masagrator,
After updating I boot into Hekate and always do "Fix Archive Bit" before launching the fusee payload. In the past archive bits have been changed after running this procedure in Hekate, this past time it said everything was as it should be/nothing changed.
There is a possibility in the past I neglected to do this. Would the Hekate bit fixer always get everything back to where it should be?

Tendiemancome, 2023-02-12:
In the past, I've used a program call "switchSD-macOS" for troubleshooting when I thought the archive bit was an issue. 
https://github.com/Player-0ne/switchSD-macOS

Tendiemancome, 2023-02-12:
I just used a windows computer to verify that archive bit is not set for every file and folder on the SD card, except for all the files and folders under "/Nintendo/Contents/registered/". I did not verify whether archive bit was set of unset in any of the files/folders within "/Nintendo/Contents/registered/".

SciresM, 2023-02-12:
`fs::ResultPortSdCardDeviceStatusDeviceEccFailed()`

That's exotic.

The error is on attempting to access the automatic backup file; fix your archive bits, if this doesn't work, it's likely an sd card error.

closed 2023-02-12 by SciresM

Tendiemancome, 2023-02-13:
Hi SciresM,
Thanks for again for getting back to me.
Its very nice to know something more specific about this error :) Will double check the archive bit again.