Atmosphere-NX/Atmosphere #2036: 16.0.0 Support

SciresM, 2023-02-21:
Master issue/check list of stuff to do for 16.0.0 changes.

- [ ] Reverse engineer new kernel changes 
- [ ] Determine what system modules need reverse engineering changes
- [ ] Update fusee/exosphere with new keydata/values
- [ ] Implement new changes into mesosphere
- [ ] Implement new changes into <sysmodules, create as needed>
- [ ] Update emummc
- [ ] Miscellaneous exosphere/fusee fixes for new DRAM IDs
- [ ] Make sure libnx is updated for anything relevant.

Post-release:
- [ ] Reverse engineer potential new boot sysmodule changes.
- [ ] Miscellaneous post-release reverse engineering
- [ ] Miscellaneous post-release implementation work/updating

appleforever11, 2023-02-21:
I accidentally updated my switch to 16.0, so am unfortunately waiting for the new Atmosphere. Any idea on a timeframe and or ETA? Thank you!

SciresM, 2023-02-21:
Issue locked, to keep it purely as a source of status.

closed 2023-10-27 by SciresM