Atmosphere-NX/Atmosphere #2040: 1.5.0 Prerelease bug report

ElementalTJ, 2023-02-23:
## Bug Report
</br>
### What's the issue you encountered?

</br>
Booting
</br>
Updated to Atmosphere 1.5.0
</br>
Updated switch to 16.0.0
</br>
### How can the issue be reproduced?
</br>
Use SX RCM Loader to boot Hekate.
Load fusee.bin
See Atmosphere logo, see Switch logo.
Crash
</br>
### Crash Report

</br>
https://drive.google.com/file/d/1ft0JECzz5sXEPRgWxS8EAtIqv0sg6odI/view?usp=sharing
(report_0000000009417712.bin found in fatal_error folder)
</br>

### System Firmware Version

16.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
fusee.bin using hekate

- Official release or unofficial build: 
  prerelease build
- Do you have additional kips or sysmodules you're loading:
none
- Homebrew software installed: [ * ]
'HATS'
- EmuMMC or SysNAND:
SysNAND, FAT32 

### Additional context?

- Additional info about your environment:
Hope this helps development. Thanks!

shae23, 2023-02-23:
010000000000bd00 is the folder for Mission Control located in atmosphere/contents. Delete that folder and you should be good. Though it goes to say, Mission Control will no longer work.

closed 2023-02-23 by SciresM

BKING1992, 2023-02-23:
[image]

masagrator, 2023-02-23:
All tesla overlays require update.

This will solve your crash issue at boot 
https://github.com/ELY3M/Tesla-Menu/releases/

BKING1992, 2023-02-23:
> masagrator

Thanks , solve the first problem . but get another error came again.
[image]

BKING1992, 2023-02-23:
> > masagrator
> 
> Thanks , solve the first problem . but get another error came again. [image]

solve the problem after deleted 0100000000001000 folder , working fine now. thanks only themes missing for now