Atmosphere-NX/Atmosphere #2043: Atmosphere 1.5.0 Pre-Release semi-boots then black screens

RyokoAsakura, 2023-02-24:
## Bug Report
### What's the issue you encountered?
Atmosphere 1.5.0 Pre-Release will not boot to the desktop once installed.

### How can the issue be reproduced?
Open Atmosphere 1.5.0 Pre-Release zip, copy to PC mounted SD card (using overwrite current files selection), replace card in Switch, use TegraRCMGUI to payload using Fusee 1.5.0pre.

Switch then displays Atmosphere logo, followed by battery icon, followed by black screen. No other activity noted.

### Crash Report
Last crash report with nearest time to issue:
https://gist.github.com/RyokoAsakura/029d83877819f503433a73a5b8aed14a

### System Firmware Version
15.0.0

### Environment?
- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
Fusee 1.5.0pre
- Official release or unofficial build: 
Official Release 1.5.0pre
- Homebrew software installed:
Too much to list, the main baddies are usually Themes (1000) and Mission Control (bd00).
- EmuMMC or SysNAND:
EmuMMC (Partition Based)

SciresM, 2023-02-24:
Does the issue occur with no homebrew (empty atmosphere/contents)? There's also some emummc issue that causes black screen for some people on partition -- this may be a duplicate of #1983. If it is, switching to file based emummc would solve.

RyokoAsakura, 2023-02-24:
Booting with a completely empty contents only slightly changes the behavior. The boot goes Atmosphere screen, Atmosphere logo, then black screen. 

Reading #1983 it could be that. I have a working backup of last known, so I'm going to attempt a new file-based EmuMMC and will report back the results.

Thanks for your help and for all of *waves arms around* this. :)

RyokoAsakura, 2023-02-24:
Moved to file-based EmuMMC and fresh loadout and it works perfectly, so I am going to close this ticket unresolved. Note to others for the future, use file based EmuMMC and take frequent full backups of your SD card.

Thanks SciresM for all your help.

closed 2023-02-24 by RyokoAsakura