Atmosphere-NX/Atmosphere #2044: 1.5.0 pre building error

impeeza, 2023-02-24:
## Bug Report

### What's the issue you encountered?

When you try to building the repo using MSYS or Linux Ubuntu the repo (commit 46094cf) the proces end with error.

the main error is `error: 'FsContentAttributes' in namespace '::' does not name a type; did you mean 'padGetAttributes'`

Even try to build latest LibNX by cloning the repo and running `make install`, the problem persist.

There is any other dependency or extra step needed to build the latest Atmosphère commit?

Thanks a lot for your work

SciresM, 2023-02-24:
Atmosphere sometimes (especially after system updates) requires latest dev branch from the Atmosphere-NX/libnx fork, which is currently PR'd to libnx master (and a work in progress).

closed 2023-02-24 by SciresM

impeeza, 2023-02-24:
Thank you so much. Regards from Colombia