Atmosphere-NX/Atmosphere #2047: 1.5.0 Pre-release

PixieDustDGE, 2023-02-25:
Gets stuck on Switch logo screen when booting. Attempted to put in a clean atmosphere folder and it does the same thing. Copying files from sd card mounted on pc then using tegrarcmgui to launch.

Left-Humor, 2023-02-25:
So when can release?

Left-Humor, 2023-02-25:
So when can it be officially released?

XModxGodX, 2023-02-25:
I had the same issue but figured out that I needed to delete some outdated folders that effect system files in content and possibly some exefs patches for the system menu. Basically delete any folder in 'contents' that you don't recognize the contents of.

PixieDustDGE, 2023-02-25:
> I had the same issue but figured out that I needed to delete some outdated folders that effect system files in content and possibly some exefs patches for the system menu. Basically delete any folder in 'contents' that you don't recognize the contents of.

I just deleted contents. The problem persists. Unless there's something I need in contents?

XModxGodX, 2023-02-25:
You might need to delete 'exefs_patches' as well if you have one.

PixieDustDGE, 2023-02-25:
I have deleted both and still have the problem...

XModxGodX, 2023-02-25:
Only other solution I can think of now is to delete the entire 'Atmosphere' folder and then copy the folder with the same name from the prerelease.

PixieDustDGE, 2023-02-25:
I'm screwed

ELY3M, 2023-02-25:
I only have sysDVR / sys-ftpd / tesla / sysbot-base  in my contents   I am fully updated to 16  
atmosphere have been slow booting and I was stuck on switch logo screen for like one min or so.  
on my other switch fully updated to 16 too - same setup but fast booting up.

ELY3M, 2023-02-25:
might wait for few mins and see if atmosphere boot up.

ELY3M, 2023-02-25:
I am using emummc as SD partition.   are you using emummc?

Masamune3210, 2023-02-25:
This isn't a support forum, nor is it a chat thread. Please go figure out your issues somewhere else. Once you narrow down the issue, you can follow the steps to make a issue report

SciresM, 2023-02-25:
You didn't use the issue templates, also.

closed 2023-02-25 by SciresM