Atmosphere-NX/Atmosphere #2048: Atmosphere crashes when installing GBA NSO through DBI

gelson-zord, 2023-02-25:
## Bug Report

### What's the issue you encountered?
Atmosphere crashes when installing GBA NSO through DBI. </br>
[crash](https://i.imgur.com/fM1Tkn7.jpeg)

### How can the issue be reproduced?
On DBI using option 'Run MTP responder' and copying the .nsp file through USB cable to the 'Micro SD Install' inside Switch, on Windows Explorer. </br>
1) plug Switch to the computer with a USB cable
2) open DBI homebrew
3) select Run MTP responder. It will open Switch intertal storage on Windows Explorer, like a smartphone
4) copy the .nsp file to the MicroSD install storage
5) Atmospheres crashes!

### Crash Report
[gist](https://gist.github.com/gelson-zord/cb9e097919f6d00364c23deca1b55695)

### System Firmware Version
16.0.0</br>

### Environment?
hekate v6.0.2 & Nyx v1.5.2 official release

- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: DBI 490, CNX Updater, Tinfoil
- EmuMMC or SysNAND: EmuMMC partition-based

duckbill007, 2023-02-25:
16.0.0 support added to DBI 499+

SciresM, 2023-02-25:
Piracy is not supported.

closed 2023-02-25 by SciresM