Atmosphere-NX/Atmosphere #2049: ERROR CODE 2001-0123(0xf601)

NinUto, 2023-02-25:
[##](url) Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

masagrator, 2023-02-25:
You did not fill out issue template.

If this happens on 16.0.0 and you used Tesla, then you need to update all tesla overlays, including Tesla Menu.

NinUto, 2023-02-25:
[image]

masagrator, 2023-02-25:
So yeah, you must update overlays. It's not an atmosphere issue.

NinUto, 2023-02-25:
> You did not fill out issue template.
> 
> If this happens on 16.0.0 and you used Tesla, then you need to update all tesla overlays, including Tesla Menu. And then it's not an atmosphere issue.

sorry my english is not good but thanks a lot

NinUto, 2023-02-25:
> So yeah, you must update overlays. It's not an atmosphere issue.

dude tesla overlays has so many "things" i really have no idea what i had use and which one i need to update, ill just search out one by one

closed 2023-02-25 by SciresM