Atmosphere-NX/Atmosphere #2050: package3 custom splash screen patcher not working

Dialgatrainer02, 2023-02-25:
## Bug Report

### What's the issue you encountered?

[before, I was changing the heakate bootlogo to use a custom one]</br>
[ no]</br>
[n/a ]</br>

### How can the issue be reproduced?

[ * ]</br>
[get a 32bit rgb bmp image and a clean package3 on ams 1.5.0 and try to patch it and boot  ]</br>

### Crash Report

the console still boots but without the splash screen and subsequent reboots use the deafult ams logo ]</br>

### System Firmware Version

</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?
16.0
- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:heakate
- Official release or unofficial build: offical pre-release from github releases
  - [ Official release version x.x.x (or) unofficial build ]
- Do you have additional kips or sysmodules you're loading:custom bootlogo and no sig checks
- Homebrew software installed: [ n/a]
- EmuMMC or SysNAND: applys to both
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

i have tried different bmp format anything other than 32 bit rgba will cause the script to crash its the same bmp is used for heakate bootscreen so i know it works

SciresM, 2023-02-25:
This is definitely a support issue; the script works. Seek help on discord

closed 2023-02-25 by SciresM

Hxyspace, 2023-10-15:
I encountered the same issue on ams 1.5.5

SciresM, 2023-10-15:
There is no issue here.