Atmosphere-NX/Atmosphere #2052: Atmosphere 1.5.0

VraelNocturnal, 2023-02-26:
On initial start up, with a wiped SD card, I will immediately get an error code 2001-0123 (0xf601) in program 420000000007E51A, This is on firmware 16.0.0, (Atmosphere 1.5.0-master-356d89244).

If I try with Hekate, and then fusee I get "A fatal error occured when running Fusse. Unable to identify package1!"

masagrator, 2023-02-26:
1. You did not use issue template
2. You did not update Tesla Menu
3. you did not use correct fusee version

VraelNocturnal, 2023-02-26:
My apologies for not using the template.

The tesla menu can't be an issue, as I'm using the new version of Atmosphere on an entirely wiped (bar the Nintendo folder) SD card.

I'm using the latest release of Hekate (6.0.2, directly from their Github) I can't see how Fusee would be mismatched but I'll reinstall and try again with it.

masagrator, 2023-02-26:
420000000007E51A is a titleid of sysmodule responsible for Tesla overlays.

SciresM, 2023-02-26:
This is a Tesla problem, not an atmosphere problem.

closed 2023-02-26 by SciresM

tbogard, 2023-03-05:
wait a moment, I had in no moment installed tesla, so I have to manually remove it even in a clean SD.

Masamune3210, 2023-03-05:
Its not a clean sd or you arent using the official package. Tesla is its own project, so the only way it could have gotten onto the card is either through intention or those avenues

SciresM, 2023-03-05:
@tbogard you also aren't the person who made this issue.

Brandon98SE, 2023-03-05:
#### Information about the issue:
- It seems like this is specifically an **nx-ovlloader** compatability issue with 16.0.0, nx-ovlloader is the chainloader for Tesla overlay, this issue also occures when only Tesla overlay is present.

# Temporary Solution: (Fixed mine)
- Remove the `/atmosphere/contents/420000000007E51A` folder from your SD Card until an update has been released for nx-ovlloader, I'm gonna keep an eye on this issue: https://github.com/WerWolv/nx-ovlloader/issues/23

tbogard, 2023-03-06:
> @tbogard you also aren't the person who made this issue.

sorry if I lifted on you some blisters on this issue, but I noted that, after formatting my switch, in a clean card, passing atmosphere as is on 1.5.0 gave me the same errors again and again. I did what @Brandon98SE said and pretty much solved my problem.

Masamune3210, 2023-03-06:
This still isn't with a clean sd and still isn't a atmosphere issue. The issue has already been raised and in fact is either close to being fixed or is already fixed with a new release. Please move over to their repo if you have continued issues

SciresM, 2023-03-06:
correction: there is no issue in atmosphere. I'm just gonna lock this.