Atmosphere-NX/Atmosphere #2055: Nintendo Logo dont disappear

yokanes, 2023-02-27:
Hi! all.

When restart nintendo switch, atmosphere appear and later appear nintendo switch logo, but lock it and cant continue

thank you for all ! 

^_^

masagrator, 2023-02-28:
You didn't use issue template. Without filling it properly we can't help you.

yokanes, 2023-02-28:
wherer is the issue template to fill it?

masagrator, 2023-02-28:
You deleted it when creating Bug Report.

Issues - > New - > Bug Report

yokanes, 2023-02-28:
## Bug: Logo nintendo block switch

Copy the version re-release 1.5.0 next to your fusee.

Once atmosphere 1.5.0 is installed with its respective fuse, when the console starts the amosphere logo appears, the next "triangle" logo and the next the nintendo switch logo (which does not disappear), but at that moment nothing else happens and the console gives the following error:
    Error code; 2001-0123 (oxf60)
    Program: 420000000007E51A
    Firmware: (atmosphere 1.5.0-master-356d89244)
   Press the power button to restart...

masagrator, 2023-02-28:
You don't "write issue based on template", you must fill this template with all data that is asked, not only selected few.

masagrator, 2023-02-28:
But okay, you did not update Tesla Menu. This is not atmosphere issue

yokanes, 2023-02-28:
should i delete the atmosphere folder from my sd and paste the version 1.5.0 directly? that way the enu tesla should be corrected, right?

urherenow, 2023-02-28:
420000000007E51A

Right there in your post. Delete that folder from atmosphere/contents. Or get an updated tesla menu, which should overwrite it. And please close this, as stated already, it's not an Atmosphere issue.

yokanes, 2023-02-28:
ok thank u

masagrator, 2023-02-28:
It won't overwrite it because Tesla Menu goes to switch/.overlays

urherenow, 2023-02-28:
Ahhh… that’s right. That folder is the ovlloader. Replacing that won’t fix anything, but technically deleting the folder will. It will also disable tesla menu though…

closed 2023-02-28 by SciresM