Atmosphere-NX/Atmosphere #2057: Atmosphere Hangs At Switch Logo

PixieDustDGE, 2023-02-28:
Bug Report
What's the issue you encountered?
Atmosphere hangs at the Switch logo

How can the issue be reproduced?
Launch Atmosphere with Fusee

Crash Report
N/A

System Firmware Version
16.0.0

Environment?
Fusee, Official
SysNAND, EmuNAND File, EmuNAND Partition

Additional context?
Contents folder was deleted, replaced all necessary folders with clean versions. I've launched via TegraRcmGUI and Hekate.

SciresM, 2023-02-28:
Okay, you made a new issue, but you *didn't answer my questions*. I also don't know why you made a new issue.

PixieDustDGE, 2023-02-28:
It occurs on emummc both file and partition-based

SciresM, 2023-02-28:
And does it occur on sysmmc...? Please actually test this.

PixieDustDGE, 2023-02-28:
It does

PixieDustDGE, 2023-02-28:
I've tried both extensively

SciresM, 2023-02-28:
Okay. If you completely replace your emmc contents, using an unbrick guide (like https://suchmememanyskill.github.io/guides/unbrick/), does it work?

PixieDustDGE, 2023-02-28:
I will check

jeffyjeffy1023, 2023-03-01:
Have you tried updating Mission Control if you've installed it? That's what fixed it for me.

SciresM, 2023-03-01:
@jeffyjeffy1023 Please do not post comments when you don't really know what you're talking about.

kurainooni, 2023-03-02:
the easiest fix would be to just disable any system module, then try to boot.( which is the recommended step in the guide SciresM linked to, referring to Mid and Late boot errors) If you can boot to CFW, you can try to enable system modules one at a time and try to boot, to figure out which don't work right now. This helped in my case, since I could boot into stock firmware just fine, but I would get BSOD on CFW( "late boot" error).

SciresM, 2023-03-02:
@kurainooni Please do not post comments when you don't really know what you're talking about.

The user has no custom sysmodules, per "deleted contents folder".

urherenow, 2023-03-02:
I'd like to know how long you waited until deciding that it was frozen at the logo. Since the update, I've noticed that booting takes considerably longer than before, and the very first boot after updating emummc, it took a full couple of minutes to get to HOS. Faster after the first boot, but still longer than <16. Can you also post a screenshot of your sd info screen from nyx, please (you'll need to inject hekate for that, of course, not fusee)?

SciresM, 2023-03-02:
@urherenow Note that that's for boots into Atmosphere 1.4.1 and higher when clearing erpt reports:

  * The mechanism for automatically cleaning up `erpt_reports` added in 1.3.0 was fixed.
    * This was actually just very fundamentally broken and has never worked, but it is verified working now.
    *  **Please note**: Your first boot into 1.4.1 may take much longer than usual (showing only a black screen), if the automatic cleanup mechanism is triggered and thousands of files are deleted. This should be one-time and not more than a minute or so, if it occurs.

That should be a one-time slowdown, since no additional logic runs on the atmosphere side except on the first boot.

PixieDustDGE, 2023-03-03:
> Okay. If you completely replace your emmc contents, using an unbrick guide (like https://suchmememanyskill.github.io/guides/unbrick/), does it work?

It does not.

SciresM, 2023-03-03:
...this is including on sysmmc? That is *really* surprising and shouldn't happen.

Mr-Socket-bit, 2023-03-04:
I tried booting the switch (Sysmmc, used hekate 6.0.2), got stuck in the Switch logo screen as described here and, after waiting for a while, I got an Atmosphere crash with code 2001-0123 (0xf601). I wasn't able to replicate this on emummc partition (It just boots normally), but firmware is 15.0.1 there, while sysmmc has 16.0.0

Here's a screenshot of the full error. Hope this helps, if only a little

[image]

urherenow, 2023-03-04:
Well, that tells me that your problem is different, because your problem is your sysmodules (tesla and such) are not the updated builds for fw 16.

Waltee, 2023-03-04:
My apologies if I shouldn't be posting here, but I have over 30 consoles and 4 of them had this issue upon updating with exactly the same files as all the rest (the rest just worked perfectly). No matter how many reboots those 4 would always get stuck at the Switch logo screen. 

What fixed it for me was manually going into the atmosphere folder on the SD and manually deleting all of the crash report, error report etc. folders. 

After that every single console booted and has continued to boot flawlessly.

urherenow, 2023-03-04:
They weren’t frozen, you were just impatient. That was a one-time deal while it was going through those reports and deleting them. But the error posted above points directly to the tesla menu. Install Tesla, built for fw 16 to take care of that.

SciresM, 2023-03-04:
Please don't take over other people's issues.

closed 2023-03-05 by PixieDustDGE