Atmosphere-NX/Atmosphere #2058: Crash at switch logo, no matter what

jeffyjeffy1023, 2023-03-01:
## Bug Report

### What's the issue you encountered?

Updated atmosphere and hekate to their latest versions, (including the atmosphere pre-release) then updated my EmuMMC (partiton) to 16.0.0. Atmosphere now crashes consistently at the Nintendo Switch logo.</br>
I have not made any changes to atmosphere.</br>

### How can the issue be reproduced?

Boot switch into RCM -> Inject Hekate in there via tegrarcmgui -> Hekate loads, shows my custom little boot logo -> Boots atmosphere, shows atmosphere logo -> crash at switch logo</br>

### Crash Report

See attatched and zip'd .bin that Atmosphere spat out unto my SD card. I can't read it myself and github won't accept it raw.

### System Firmware Version

16.0.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Hekate (not sure if fusee is involved but I'm pretty sure it isn't?)
- Official release or unofficial build: 
  - Official Pre-Release 1.5.0 (sixty-ninth release... nice)

- Do you have additional kips or sysmodules you're loading:
  -No mods/sysmodules/kips are installed, nor do I have sigpatches currently.
- EmuMMC or SysNAND:
  - EmuMMC, partition based

### Additional context?

- Literally all I did was update with daybreak, reboot, launch atmosphere, then boom. Crashes every time.
- Specific error text:
Panic occurred while running Atmosphere.

Title ID: 010000000000BD00
Error:     std::abort (0xFFE)
(report path)

Press POWER to continue. (should say press power to reboot honestly, but that's not the main problem I have here is it now?)

[report_09ec1844.zip](https://github.com/Atmosphere-NX/Atmosphere/files/10856918/report_09ec1844.zip)

SciresM, 2023-03-01:
Update mission control. This isn't an atmosphere problem.

closed 2023-03-01 by SciresM

jeffyjeffy1023, 2023-03-01:
...But I don't remember ever installing mission control?

jeffyjeffy1023, 2023-03-01:
however it does turn out that I may have forgotten to update fusee.bin so I'm going to go do that and then install mission control for the hell of it (since I already love bloopair on the Wii U) once I know atmosphere is (for a pre-release at least) stable

jeffyjeffy1023, 2023-03-01:
Still crashes... what the fuck... why is mission control even a requirement here for it to NOT crash?

masagrator, 2023-03-01:
010000000000BD00 is a titleid of mission control, so you must have it. There is no other option. Delete it or update it

jeffyjeffy1023, 2023-03-01:
Well that's weird because I don't ever recall installing it, but I'll update it then. My bad I guess

jeffyjeffy1023, 2023-03-01:
yep, updating mission control fixed it. i may be stupid

thejocraftkit, 2023-04-21:
how to delete mission controll

Masamune3210, 2023-04-21:
You installed it, you should know how to remove it.

Durionleponey, 2023-06-16:
> 010000000000BD00 is a titleid of mission control, so you must have it. There is no other option. Delete it or update it

thank you so much, you've easily saved me 2 hours of my time!