Atmosphere-NX/Atmosphere #2061: software has closed" since updating FW16 on CFW and updating dlc/game

chrisssj2, 2023-03-02:
I used to run My hero justice 2 on FW 14 before updating. Not sure if I had dlc installed.
Now I updated to FW16 with the latest atmosphere+hekate. and also i (re) Installed all dlc for my hero justice 2 + latest update,

This is where the problem started happening. "software has been closed because of error" With all 41 dlc's installed.
I decided to install just a couple and the game would run, then after I try install a few more only to see again "software has been closed because of error"
I remove all DLC, and the game ran fine again. I kept retrying and seeing what DLC is causing problem. I couldn't find out. and i deleted all dlc again.

Now even base game+update has the "software has been closed because of error" And I can't even run it anymore....
What is going on, and how can I fix it...?

I tried go into maintenance mode. Try reset game version. Delete save data for the game. Check for integrity. all OK.

16.0.0 AMS 1.5.0 E Game version 1.1.5 V983040

SciresM, 2023-03-02:
Piracy is not supported, which would be why you were banned on discord.

closed 2023-03-02 by SciresM

chrisssj2, 2023-03-02:
> Piracy is not supported, which would be why you were banned on discord.

Why do you assume this? I never said such thing.

SciresM, 2023-03-02:
Feel free to DM me on discord, if you're really interested in arguing the point. `SciresM#0524`.