Atmosphere-NX/Atmosphere #2062: blue screen when starting OFW or CFW with atmosphere 1.5.0 mariko sxcore modchip

RonnyIvanFV, 2023-03-03:
## Bug Report

### What's the issue you encountered?
I have switch mariko v2 with sxcore modchip with stock firmware(teamxcuter) </br> it worked perfectly with hekate v6.0.1 bootloader and atmosphere 1.4.0 with OFW 15.0.1 / CFW15.0.1

======so my dog ​​decided to update OFW - CFW -atmoephere and hekate
as a result I got the following
blue screen after updating CFW 1.6.0- OFW1.6.0 and atmosphere1.5.0

</br>

### How can the issue be reproduced?
I am using sxcore with teamxecuter firmware
switch mariko v2
download atmosphere from official github site
download hekate from official github
download sigpatches from https://gbatemp.net/threads/sigpatches-for-atmosphere-hekate-fss0-fusee-package3.571543/

1 start OFW 15.0.1 connect to wifi and upgrade to 16.0.0
2 update hekate and atmosphe like i do regularly
 to update hekate and atmosphere I usually delete the atmosphere, bootloader, payload.bin(root) 
 copy the new atmosphere and bootloader
 copy the atmospere folder containing the sigpatches
 copy fuse.bin into payloads folder
 I rename the hekate_ctcaer_.bin file to payload.bin because my mod chip has teamxcuter firmware so it can boot hekate
3 updated hekate and atmosphere start CFW emunand
4 update CFW emu nand 15.0.1 with daybreak to 1.6.0
5 reboot my switch
6 try to start CFW emunand and it shows me blue screen
7 try to start OFW SYSnand and it shows blue screen
8 It is only possible to start OFW with the reboot OFW option in the hekate menu

### Additional context?
-I have a copy of my emunand 15.0.1 on another sd card it is functional but without the option to put the console to sleep
-I use sxcore modchip with sxos firmware

SciresM, 2023-03-03:
Blue screen indicates some hw corruption or pk21 corruption, typically. That said,

> team xecuter firmware

Not supported. Use spacecraft or similar.

Also, "sigpatches" piracy not supported.

closed 2023-03-03 by SciresM

RonnyIvanFV, 2023-03-05:
Thanks for answering, I tried again to update atmosphere and hekate with a
series of steps, create for my case and this time I can do it without
errors, the problem was that I downloaded the sigpatch from another page, I
will take your advice to change the frimware of my mod into account chip,
my console is working normally with ofw and cfw