Atmosphere-NX/Atmosphere #2064: Atmosphere 1.5.0 crashes seconds after Nintendo Switch logo

HatchlingByHeart, 2023-03-04:
**What's the issue you encountered?**
Switch boots into Atmosphère, but crashes to error screen shortly after the Nintendo Switch logo is shown.

**How can the issue be reproduced?**
Booting into CFW (SysNAND) after updating HOS to 16.0.0 and Atmosphère to 1.5.0

**Crash Report**
(/atmosphere/crash_reports did not exist and was not generated, but there was this in /atmosphere/fatal_reports)

Atmosphère Fatal Report (v1.1):
Result:                          0xF601 (2001-0123)
Program ID:                      420000000007e51a
Process Name:                    ovlloader
Firmware:                        16.0.0 (Atmosphère 1.5.0-master-356d89244)
General Purpose Registers:
Start Address:                   0000000000000000
Stack Trace:
        ReturnAddress[00]:       0000000000000000
        ReturnAddress[01]:       0000000000000000
        ReturnAddress[02]:       0000000000000000
        ReturnAddress[03]:       0000000000000000
        ReturnAddress[04]:       0000000000000000
        ReturnAddress[05]:       0000000000000000
        ReturnAddress[06]:       0000000000000000
        ReturnAddress[07]:       0000000000000000
        ReturnAddress[08]:       0000000000000000
        ReturnAddress[09]:       0000000000000000
        ReturnAddress[10]:       0000000000000000
        ReturnAddress[11]:       0000000000000000
        ReturnAddress[12]:       0000000000000000
        ReturnAddress[13]:       0000000000000000
        ReturnAddress[14]:       0000000000000000
        ReturnAddress[15]:       0000000000000000
        ReturnAddress[16]:       0000000000000000
        ReturnAddress[17]:       0000000000000000
        ReturnAddress[18]:       0000000000000000
        ReturnAddress[19]:       0000000000000000
        ReturnAddress[20]:       0000000000000000
        ReturnAddress[21]:       0000000000000000
        ReturnAddress[22]:       0000000000000000
        ReturnAddress[23]:       0000000000000000
        ReturnAddress[24]:       0000000000000000
        ReturnAddress[25]:       0000000000000000
        ReturnAddress[26]:       0000000000000000
        ReturnAddress[27]:       0000000000000000
        ReturnAddress[28]:       0000000000000000
        ReturnAddress[29]:       0000000000000000
        ReturnAddress[30]:       0000000000000000
        ReturnAddress[31]:       0000000000000000

**System Firmware Version**
16.0.0

**Environment**
Payload: Hekate
Release: Official 1.5.0
Additional kips or sysmodules: None
Homebrew software installed:
- AIO Switch Updater
- Breeze
- Daybreak
- EdiZon
- FTPD-Classic
- Goldleaf
- JKSV
- nxdumptool
- Switch 90DNS Tester
- Switch Cheats Updater

**EmuMMC or SysNAND**
SysNAND

Crash occurs even with a fresh install of Atmosphère and all software.
Not sure if relevant, but it's an Australian Switch with Australian Switch Account. Using a 512GB microSD formatted as FAT32.

I hope I did this bug report correctly.

masagrator, 2023-03-04:
You did not not update Tesla overlays to releases supporting 16.0.0, Tesla Menu included

HatchlingByHeart, 2023-03-04:
> You did not not update Tesla overlays to releases supporting 16.0.0, Tesla Menu included.
> 
> Exactly this error definitely won't show if you use fresh Atmosphere install.

I hope I overlooked this because I am certain I fresh installed everything with latest versions. I will check.

HatchlingByHeart, 2023-03-04:
Tesla overlays were already latest version, but I was able to get it to boot by outright disabling Tesla. Thanks for your help!

closed 2023-03-04 by HatchlingByHeart

miigotu, 2023-03-05:
I solved this by deleting boot2 sysmodules with  "commonproblemresolver.bin"