Atmosphere-NX/Atmosphere #2067: i need some help with my atmosphere...

ldsujtoi, 2023-03-04:
hello together,

first of all, i'm sorry for my bad english. I hope u understand me and can help me with my problem.

i have a cracked switch and it worked well, but on the updatemessage i accept the update and my switch run out of battery pack... i restarted the switch with my rcm kit, but got the message "A fatal error occured when running fusee. Unable to identify package1! i tried some restarts but the errormessage was the same...

Does anyone know the problem and has a solution where I don't have to reset the switch?

Clyxos, 2023-03-04:
You are on an older version of atmosphere, which doesn't support 16.00. Update to the pre-release.

SciresM, 2023-03-04:
The issue tracker is for bugs, not for support. And you didn't use the issue tracker.

closed 2023-03-04 by SciresM