Atmosphere-NX/Atmosphere #2068: error 2001-0123 0xf601

Minecraftplayer20, 2023-03-05:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ I got a code that said  2001-0123 0xf601 before I encountered this I was just putting in the latest files. ]</br>
[ I didn't make any changes to atmosphere ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ To get this issue you have to go to sdsetup and click on default selection then to atmosphere and get the latest version plus hekate then putting in the files then going to tegraRCM injecting you switch booting with an RCMclip booting into sysMMC it shows you the error. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/Minecraftplayer20/72cfd33a964e54a03e067f2541c05f95) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

16.0.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official
  - [ Official release version 6.0.2]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading: nope
- Homebrew software installed: [ 3 ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

masagrator, 2023-03-05:
You didn't include all necessary informations.

If titleid on crash screen is the same as here, answer is the same
https://github.com/Atmosphere-NX/Atmosphere/issues/2065

Brandon98SE, 2023-03-05:
This is an [nx-ovlloader/Tesla](https://github.com/WerWolv/nx-ovlloader) issue, [here is a temporary workaround](url) until it has been updated.

Minecraftplayer20, 2023-03-06:
> This is an [nx-ovlloader/Tesla](https://github.com/WerWolv/nx-ovlloader) issue, [here is a temporary workaround](url) until it has been updated.

Where do I put this file?

closed 2023-03-06 by SciresM