Atmosphere-NX/Atmosphere #2069: Time system does not work correctly

manolo754, 2023-03-06:
Time system does not work correctly

I started Twitch app and the app has a time certificate error.
You can be reproduced error starting Twitch app.

System Firmware Version: 16.0.0
Atmosphere version: 1.5.0 pre-release
Bootloader: hekate v6.0.2
   Do you have additional kips or sysmodules you're loading: yes
- Homebrew software installed: [ some applicattions ]
- EmuMMC partition-based

masagrator, 2023-03-06:
> Do you have additional kips or sysmodules you're loading: yes

That's actually pretty important to list them all. Atmosphere doesn't touch time sysmodule, so whatever happens should not be related to Atmosphere itself.
And question - does it work with OFW correctly?

Masamune3210, 2023-03-07:
There is a reason the template is there, generalizing what kips and homebrew you have as "yes" and "some applications" helps absolutely no one

ELY3M, 2023-03-07:
I have zero issues with twitch and I am online in atmosphere.  My switch clock is always synced.   
I downloaded twitch from e-shop as normally.

happywolf, 2023-03-08:
> 
I'm surprised you manage to get 1.5.0 pre-release working at all o.o

happywolf, 2023-03-08:
> Time system does not work correctly
> 
> I started Twitch app and the app has a time certificate error. You can be reproduced error starting Twitch app. If you have time system automatic, the hour and the date it is not correct. System Firmware Version: 16.0.0 Atmosphere version: 1.5.0 pre-release Bootloader: hekate v6.0.2 Do you have additional kips or sysmodules you're loading: yes
> 
>     * Homebrew software installed: [ some applicattions ]
> 
>     * EmuMMC partition-based

I'm surprised you even got the pre-release to work at all :o

SciresM, 2023-03-08:
Closing this because of lack of info.

closed 2023-03-08 by SciresM