Atmosphere-NX/Atmosphere #2079: Error 2002-3502 then EmuNAND takes a long time to boot

nity0814, 2023-03-18:
## Bug Report

### What's the issue you encountered?

I got the error 2002-3502 then EmuNAND won't even boot

### How can the issue be reproduced?

by booting EmuNAND

### Crash Report

https://gist.github.com/nity0814/4deb3ee0a7f3bc6441c15c8ad1bcdd6c
### System Firmware Version

13.2.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe lauched by hekate (i added fusee.bin in hekate_ipl.bin)
- 
- Official release or unofficial build: Official build 
  - AMS 1.5.1
- Do you have additional kips or sysmodules you're loading:  No, I don't think so
- 
- Homebrew software installed: Browser, Daybreak, reboot to payload, tinwoo, tinfoil, nxmtp, goldleaf, ftpd classic, ftpd, homebrew app store
- [image]
- EmuMMC or SysNAND: EmuMMC
  - Partition BASED

nity0814, 2023-03-18:
if you wonder i used tinfoil to reinstall a corrupted game
( i do have payed the game)

Masamune3210, 2023-03-18:
Remove sigpatches and all disallowed homebrew and retest. Information tainted by disallowed modifications is sullied.

nity0814, 2023-03-18:
will it be alright by formating it again and put it again?

nity0814, 2023-03-18:
after this error my sd card boots up in 1bit mode on hekate

TTghostsemir, 2023-03-20:
Piracy is not supported!
@SciresM Piracy is here.

And about the 1-bit errors, it's usually the MicroSD card connector or the MicroSD card is broken.

closed 2023-03-20 by SciresM