Atmosphere-NX/Atmosphere #2083: Atmosphere crashes after reopening suspended game

Xarber, 2023-03-25:
System Version: 1.5.1 | 16.0.1 | E
Steps to replicate: 

1. Open any game
2. Go to the home menu
3. Reopen the game
4. Atmosphere crashes
PS: Also opening the homebrew menu from the images section crashes the switch (Usually with a game open)

andrea-ap, 2023-03-25:
I try and not have issue. System Version: 1.5.1 | 16.0.1 | E

Xarber, 2023-03-25:
> I try and not have issue. System Version: 1.5.1 | 16.0.1 | E

16.0.0 sorry, anyways i dont know how to solve and i cant find anything online

masagrator, 2023-03-25:
You did not use issue template.

Xarber, 2023-03-25:
> You did not use issue template.

i dont know how to use them and where to find them

SciresM, 2023-03-25:
I am closing. Create new issue. It will have a template in the text box. Fill it out.

closed 2023-03-25 by SciresM