Atmosphere-NX/Atmosphere #2084: Atmosphere freezing on game restart / HBMenu launch

Xarber, 2023-03-25:
## Bug Report

### What's the issue you encountered?
I did not make any changes to atmosphere, when i open any kind of game and go back to the home menu (suspending it), as soon as i open that game again atmosphere suddently freezes. This does not happen every time to me but it's very often and i need to force a shutdown and re-inject the payload to restart. The same problem occurs when im running a game and i try to open the hbmenu (i can open it holding R on a game).

### How can the issue be reproduced?
For the game freeze:
1. Open any game that uses a bit of ram
2. Go to the home menu
3. Reopen the game (Without closing it the first time)
4. The system should freeze.

For the hbmenu:
1. Open the album
2. The system should freeze.

### Crash Report
The system did not crash (no atmosphere error screen appeared), it just freezed.

### System Firmware Version
16.0.0

### Environment?
- hekate 6.0.2
- Partition-Based EmuMMC
- Atmosphere Official release 1.5.1
- Sysmodules:
  1. "Tesla"
  2. "sys-con"
  3. "00FF0000636C6BFF"
  4. "4200000000001312"
  5. "0100000000000352"
  6. "MissionControl"
  7. "SysDVR"
  8. "sys-ftpd-10k" (OFF)
  9. "Noexes" (OFF)
- Homebrew software installed: 
  1. Breeze
  2. All-In-One Switch Updater
  3. Checkpoint
  4. Edizon
  5. Edizon SE
  6. ftpd pro 3.1.0
  7. HB App Store
  8. emuiibo
  9. rit
  10. AmiiboGenerator
  11. Amiigo
  12. Daybreak
  13. DBI
  14. Friday Night Funkin' Rewritten
  15. Goldleaf
  16. ldnmitm cfg
  17. NX-Shell
  18. nxdumptool
  19. NXThemes Installer
  20. Reboot to Hekate
  21. Reboot to Payload
  22. RetroArch
  23. sys-clk manager
  24. SysDVR Settings
  25. Themezer-NX

SciresM, 2023-03-25:
> "did not make any changes to atmosphere"

You have a lot of custom sysmodules. It is going to be one of them. Also, "OFF" isn't a thing.

Please confirm the issue does not occur with all sysmodules removed? (You can temporarily rename "contents" to "contents.old", to test).

Xarber, 2023-03-25:
> > "did not make any changes to atmosphere"
> 
> You have a lot of custom sysmodules. It is going to be one of them. Also, "OFF" isn't a thing.
> 
> Please confirm the issue does not occur with all sysmodules removed? (You can temporarily rename "contents" to "contents.old", to test).

I can indeed turn off sysmodules from Edizon SE.
Anyways, after renaming the folder to contents.old, it seems to work, but i would really like to know what was the sysmodule that caused the error, is there any way to read the logs and understand them or just see which sysmodule was using a lot of ram?

SciresM, 2023-03-25:
"I can indeed turn off sysmodules from Edizon SE." this doesn't actually turn them off. Off isn't a thing.

" i would really like to know what was the sysmodule that caused the error, is there any way to read the logs and understand them or just see which sysmodule was using a lot of ram" <-- there aren't really logs. It's probably also not one sysmodule in particular, but several in concert pushing you over a threshold with no single one responsible.

If you want, you can always test one by one, to see what combinations succeed and don't, and find a combination that has the features you want without the crashing?

closed 2023-03-25 by SciresM