Atmosphere-NX/Atmosphere #2085: Black Screen on 1.5.1

brettsmods, 2023-03-28:
## Bug Report

### What's the issue you encountered?

Getting the two Atmosphere splash screens, and then a black screen. Removed entire contents folder and issue persists on 1.5.1. Downgrading to Atmosphere 1.4.0 restores functionality.

### System Firmware Version

sysMMC: 16.0.1
emuMMC: 15.0.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate 6.0.2
- Do you have additional kips or sysmodules you're loading: All disabled for now
- EmuMMC or SysNAND:
  Partition based emuMMC

iCanuck, 2023-03-28:
This was happening to me on 1.4. I realized it was freezing at the boot stage when it was clearing the erpt_reports. The fix was to find out why so many erpt_reports were being created.

brettsmods, 2023-03-28:
> This was happening to me on 1.4. I realized it was freezing at the boot stage when it was clearing the erpt_reports. The fix was to find out why so many erpt_reports were being created. You could try to use the system ini to disable the erpt_report clearing.

I disabled the erpt_report clearing, but it seems to be the same issue. Didn't seem to have a huge number of reports either.

iCanuck, 2023-03-28:
hmm sorry I can't be more help. I am not having this same issue.

SciresM, 2023-03-28:
@brettsmods Does the issue occur on sysmmc? Can you post a screenshot of the completely replaced atmosphere folder, and the contents folder if one exists?

brettsmods, 2023-03-28:
sysmmc is fine.

So I updated emummc to 16.0.1, booted and everything was sluggish with stock atmosphere folder from github zip. Swapped back to my original atmosphere folder and now it's working fine and is super quick, like it was before.

I really don't know what caused the issue of it being so slow/freezing up, but it seems to be working fine now on 1.5.1/16.0.1 and my original atmosphere and contents folders... Super weird.

SciresM, 2023-03-28:
Probably some kind of SD issue, to be honest. Glad it's resolved though I guess!

closed 2023-03-28 by SciresM