Atmosphere-NX/Atmosphere #2086: Latest version installation error/crash

JarlTheKing, 2023-03-30:
### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]
Updated to the latest Atmosphere for the latest version of the Switch. After going activating with the PC, and going through the Atmosphere loading screens before the home screen it crashes and gives a error report.

[ Did you make any changes related to Atmosphère itself? ]
Just installed the latest version over the other.

[ If so, make sure to include details relating to what exactly you changed. ]
Updated the files given in the latest Atmosphere package.

### System Firmware Version

16.0.1

### Environment?

- What bootloader: (fusèe) was Atmosphère launched by:
- 
- Official release or unofficial build: Official
  - [ Official release version 1.5.1]]

[image]

[01679712494_420000000007e51a.log](https://github.com/Atmosphere-NX/Atmosphere/files/11107219/01679712494_420000000007e51a.log)

SciresM, 2023-03-30:
Not an atmosphere issue. Update or remove Tesla and possibly other custom sysmodules.

closed 2023-03-30 by SciresM

JarlTheKing, 2023-03-31:
What's a Tesla file? Maybe also like removing the Switch Background Saver? What other files would I have to remove, again don't know a whole lot besides the basics.

SciresM, 2023-03-31:
If you don't know what you're doing, just delete the entire atmosphere dir and replace with latest release zip. You can re-install add-ons afterwards.

Also, this issue tracker is for bugs, not for support, so I'm locking this.