Atmosphere-NX/Atmosphere #2088: Mirror wifi database to SD

benwa, 2023-04-03:
## Feature Request

### What feature are you suggesting?
#### Overview:
Similar to the recently added Bluetooth database feature, it would be useful to have the Wifi database mirrored to the SD card for other applications to use.

#### Smaller Details: 
Unknown

#### Nature of Request:  
- Addition
  A file under `/atmosphere/wireless.db`
  A system setting to enable the feature

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>

### Why would this feature be useful?
This would possibly allow users who use alternate operating systems to have the same wireless networks without having to connect to them in each OS.

ndeadly, 2023-04-10:
Conceptually it would be nearly identical to my Bluetooth PR but for setsysGetNetworkSettings/setsysSetNetworkSettings. It's more a question of whether @SciresM would accept such a feature. I imagine the argument for syncing wifi information is much weaker than for Bluetooth given how infrequently one would typically update their wifi settings. Additionally. many people running emummc explicitly _don't_ want their wifi settings synced. A simple homebrew to dump the current pairing information may be considered more practical.

SciresM, 2023-04-10:
I would be willing to look at a pull request.

closed 2023-04-10 by SciresM