Atmosphere-NX/Atmosphere #2089: Insane Battery Draw and Console Heating in Sleep

mountainaught, 2023-04-04:
## Bug Report

### What's the issue you encountered?

While in sleep mode, the console both draws battery and heats up like mad. The issue almost always happens while a game is open, but only randomly happens when nothing is.

### How can the issue be reproduced?

Open up any game, let it load, then suspend console by pressing on power button and leave for a hour or two.
Upon return, the switch with have obvious load heat and a battery 25% lower than before.

Only modifications to system are DNS MITM enabled and themes applied.

### Crash Report

No crashes to report

### System Firmware Version

16.0.0 | AMS 1.5.1

### Environment?

Hekate 6.0.2
Official (bootloader and atmosphere from github)
MissionControl sysmodule
EmuMMC, Partition based

### Additional context?

- Additional info about your environment:

Issue first began after installing 16.0.0 with AMS 1.5 pre release.
Continues on 1.5.1

SciresM, 2023-04-04:
Have you verified whether the issue occurs on stock and/or on sysmmc with no modifications (like no mission control)?

mountainaught, 2023-04-06:
This is why it's always a good idea to call tech support I guess. Toggling all modifications on and off, I reached this conclusion:

The behavior was only present with dns_mitm enabled.

SciresM, 2023-04-06:
"dns mitm enabled" Do you have a custom configuration?
Could you post your configuration, if so? Do problems occur with the default configuration?

mountainaught, 2023-04-06:
https://pastebin.mozilla.org/Xd5K3Gs2
^ DNS configuration (default.txt)
I didn't try with default, since I didn't want to risk my console connecting anywhere

SciresM, 2023-04-06:
That snippet no longer exists.

SciresM, 2023-04-06:
Could you please upload that to a host where it won't be deleted before I can look at it?

mountainaught, 2023-04-08:
My apologies, set it to the wrong expiration time
https://pastebin.mozilla.org/3mBBHTte

dong2dong3, 2023-05-02:
[image]
is this normal or my battery is dead? 16.0.2

thematica, 2023-05-05:
this issue happening to me right now
and I don't think I use any custom modification

can anyone explain what is dns_mitm? I suppose I need to disable it to fix the draining issue?

thanks

orio311, 2023-08-15:
@mountainaught Did you ever find a solution other than disabling DNS_MITM? I'm facing the same issue but can't afford to risk a ban

IISAKNIFE, 2023-09-13:
I am having this issue as well. I am running 16.1.0 on Emummc with the latest Atmosphere on my OLED with HWFLY (but it has been happening for many versions of Horizon and Atmosphere).

The switch gets warm or even hot to the touch, and the battery is significantly lower than when stock, or using Atmosphere with airplane mode on. I used an older configuration of default.txt with many specific addresses before, but switched to the config using wildcards that is on the rentry guide recently, but it didn't make a difference. 

I have a modded v1 as well, and the issue has not been there (or at least has not been as noticable). The v1 is modded via jig, and is modded on sysmmc rather than emummc.

SciresM, 2025-04-03:
I think this is literally just an interaction between Nintendo's networking code and sleep, when it believes it should be able to access certain domains but can't because of dns redirections.

closed 2025-04-03 by SciresM

IISAKNIFE, 2025-04-05:
> I think this is literally just an interaction between Nintendo's networking code and sleep, when it believes it should be able to access certain domains but can't because of dns redirections.

The issue only happens on my OLED though, I have a v1 that does not have this issue at all