Atmosphere-NX/Atmosphere #2090: Issue with Atmosphere when internal memory and SD card are full

andersonbisconsin, 2023-04-05:
Description:
I found a bug in Atmosphere when opening the system. The problem seems to occur when the internal memory and SD card are full. When this happens, the system fails to start up and remains stuck at the initial Atmosphere logo.

Steps to reproduce:

Fill up the internal memory and SD card until they are both full
Attempt to start the Atmosphere system
The system fails to start and remains stuck at the initial Atmosphere logo
Expected behavior:
The system should be able to start up, regardless of the amount of memory being used.

Actual behavior:
The system fails to start up and remains stuck at the initial Atmosphere logo when the internal memory and SD card are full.

Environment:

Atmosphere version: 1.5.1
Switch firmware version: 16

SciresM, 2023-04-06:
Have you considered not having a full SD card? I doubt that's actually atmosphere and not...a Nintendo component atmosphere is relying on.

andersonbisconsin, 2023-04-06:
I did a test without filling up the memory and the problem still occurred. It might be a problem with my SD card, but I found it strange that the console became very slow before the problem happened. It seems like some processing is being done and then the system stops booting, it could be some sort of system rearrangement.

Is there a way to see what is on the Emunand internal memory on the PC?

kabihuni, 2023-04-07:
erpt_reports...clean...

closed 2023-04-19 by andersonbisconsin